bernhard-42/vscode-ocp-cad-viewer #3: SystemError: initialization of _internal failed without raising an exception

josephbejjani, 2023-04-24:
Hello, 
I successfully installed the OCP CAD Viewer. I am using CadQuery 2 and would like to display my 3D geometry in VS Code. I am receiving the following error when anything, even the example provided within the extension. Any help is appreciated!
Thanks, Joseph
---------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\Users\user\Documents\Code\CadQuery\Compressor JB\paraturbo-cad\COMP\Main Function.py", line 5, in <module>
    from ocp_vscode import *
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\ocp_vscode\__init__.py", line 17, in <module>
    from .show import (
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\ocp_vscode\show.py", line 17, in <module>
    from ocp_tessellate import PartGroup
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\ocp_tessellate\__init__.py", line 21, in <module>
    from .cad_objects import (
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\ocp_tessellate\cad_objects.py", line 20, in <module>
    from .ocp_utils import (
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\ocp_tessellate\ocp_utils.py", line 71, in <module>
    from quaternion import rotate_vectors
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\quaternion\__init__.py", line 31, in <module>
    from .quaternion_time_series import (
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\quaternion\quaternion_time_series.py", line 10, in <module>
    from quaternion.numba_wrapper import njit
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\quaternion\numba_wrapper.py", line 13, in <module>
    from numba import njit, jit, vectorize, int64, float64, complex128
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\numba\__init__.py", line 42, in <module>
    from numba.np.ufunc import (vectorize, guvectorize, threading_layer,
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\numba\np\ufunc\__init__.py", line 3, in <module>
    from numba.np.ufunc.decorators import Vectorize, GUVectorize, vectorize, guvectorize
  File "C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\numba\np\ufunc\decorators.py", line 3, in <module>
    from numba.np.ufunc import _internal
SystemError: initialization of _internal failed without raising an exception

josephbejjani, 2023-04-24:
I downgraded my version of numpy with no success. Same error displayed again.

bernhard-42, 2023-04-24:
Could you please try without numba? 
Given that OCP is a C++ library, I would have expected that it doesn't work with numba. 
So I never tested it

josephbejjani, 2023-04-24:
Thank you for the prompt reply. How can I try without numba?
I import the following in my code: 

```
import cadquery as cq
import numpy as np
import pandas as pd
from cadquery.selectors import AreaNthSelector
import pickle
from cadquery import *
from cq_warehouse import *
from cq_warehouse.extensions import *
import os
```
I need numpy to evaluate a parametric function that returns coordinate points which I then plot in 3D space. 
Thanks a lot!

josephbejjani, 2023-04-24:
```
import cadquery as cq
from Impeller import *

from ocp_vscode import *
set_port(3939)

reset_show()
set_defaults(axes=True, transparent=False, collapse=1, grid=(True, True, True))

Imp = Impeller()
Hub = Imp.hub()
show_object(Hub,options={"alpha":0, "color": (255,0,0)})
```

This is the code where I call my function. I tried it in cq-editor and it works. I would like to run it in vs code though.

josephbejjani, 2023-04-24:
I tried with a simple box. It did not work as well. 
```
import cadquery as cq
from ocp_vscode import show, show_object, reset_show, set_port, set_defaults, get_defaults
set_port(3939)

result = cq.Workplane("front").box(2.0, 2.0, 0.5)

show_object(result)
```
Small note: I never used numba in my work.

bernhard-42, 2023-04-24:
`numpy` is OK , but I see this in the stack trace: `C:\ProgramData\Anaconda3\envs\turbocomp\lib\site-packages\numba\np\ufunc_init_.py` so something installed numba ...
Googling for that talks about dependency issues

I just tried after installing numba, your last example still works.

Can you create another fresh environment to test the last example? Looks like some dependencies are maybe broken ...

josephbejjani, 2023-04-27:
Thanks a lot! It works. I tried it with several examples. However, for some of them I am receiving this message: 

```
+Traceback (most recent call last):
  File "c:\Users\user\Documents\Code\CadQuery\Compressor JB\paraturbo-cad\Impeller Calling Functions.py", line 37, in <module>
    show_object(Mainblades)
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_vscode\show.py", line 502, in show_object
    show(
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_vscode\show.py", line 335, in show
    data = _convert(
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_vscode\show.py", line 157, in _convert
    instances, shapes, states, config, count_shapes = _tessellate(
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_vscode\show.py", line 53, in _tessellate
    part_group = to_assembly(
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_tessellate\convert.py", line 806, in to_assembly
    pg, instances = _to_assembly(
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_tessellate\convert.py", line 519, in _to_assembly
    part = get_instance(cad_obj.obj, pg.name, rgba, instances, progress)
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_tessellate\convert.py", line 369, in get_instance
    obj, loc = relocate(obj)
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_tessellate\convert.py", line 408, in relocate
    loc = get_location(obj)
  File "C:\ProgramData\Anaconda3\envs\vs-cad2\lib\site-packages\ocp_tessellate\ocp_utils.py", line 768, in get_location
    raise TypeError(f"Unknown location typ {type(loc)}")
TypeError: Unknown location typ <class 'method'>
```
Any insights about this problem?

bernhard-42, 2023-04-27:
Could you maybe share the code where it happens?

bernhard-42, 2023-05-07:
@josephbejjani Is this still an issue, or can we close it?

josephbejjani, 2023-05-07:
Hello. Sorry for the delay and unresponsiveness as I was experimenting with the extension! I couldn't share my code because it is part of a confidential project at the lab I work in. The extension works for when calling some methods in the code, but does not for complicated assemblies with unions, so I am calling those heavy methods in cq-editor instead. Issue can be closed! Thanks again for the support, JB

bernhard-42, 2023-05-07:
hmm, would be nevertheless great to understand what doesn't work.
Could you build a simple example not related to your project?
It looks like you assign a method where a location is expected.

You seem to have a cadquery assembly.
The OCP CAD Viewer code then tries to identify instances (references) to ensure each object is only tessellated once.
Therefore it relocates the object to the origin and adds the original location to it separately. This is a critical step and it doesn't expect a method passed instead of a `Location` or `TopLoc_Location`.

What is your use case for passing a method instead of a location?

josephbejjani, 2023-05-10:
`assembly.add(blade[i+1],color=cq.Color('red3'), name = bladename + ' ' + str(i+2))`

I think my issue lies in such a command which I run within a loop. Sometimes it works, sometimes not.
I run a similar command while defining location but with no naming and it works fine. 
In any case, this small issue is not hindering my work. Thanks for the follow up!
JB

josephbejjani, 2023-05-10:
The goal behind this is to name the components so they appear in a CAD software table/tree with their respective names instead of long series of numbers and letters. It indeed works when I open my step in a CAD software (I tried with 2 softwares).

bernhard-42, 2023-12-12:
Since I don't know what to do here, I'll close this issue.
Please re-open if it happens to you again

closed 2023-12-12 by bernhard-42