bernhard-42/vscode-ocp-cad-viewer #5: use default color or alpha =1 if too few parameters are provided to show

bernhard-42, 2023-04-24:


bernhard-42, 2023-04-25:
agreed, I have implemented it like this.
This btw also alliws to provide colormaps, e.g. from matplotlib. Just experimenting with  it

JonnyHaystack, 2023-04-25:
Another random thought: would be neat to be able to configure the default colours for individual types of shape. E.g. if I want solids to be orange, sketches to be green, lines to be red

bernhard-42, 2023-04-27:
> Another random thought: would be neat to be able to configure the default colours for individual types of shape. E.g. if I want solids to be orange, sketches to be green, lines to be red

good point, does exist currently

bernhard-42, 2023-05-07:
@JonnyHaystack 
Opened another issue for your "random thought"

bernhard-42, 2023-05-07:
done, rest tracked in the separate issue

closed 2023-05-07 by bernhard-42