bernhard-42/vscode-ocp-cad-viewer #6: Expose parameter to allow setting explode=True

JonnyHaystack, 2023-04-29:
Should be able to set `explode=True` in `set_defaults()` or `show()` so you don't have to keep enabling the explode checkbox after each render

bernhard-42, 2023-05-07:
@JonnyHaystack There is too much interaction with the animation part to add it to `set_defaults`.
It will only be available as a parameter to `show`, but **not**  as parameter to `show_object` (doesn't make sense to explode one object).
Would that work?

bernhard-42, 2023-05-07:
Ha, found a way to control the dependency between explode and animate.
The explode setting will be available in VS Code Workspace config, with `set_defaults` and with `show`.
Just not with `show_object`

bernhard-42, 2023-05-07:
Fixed in 1.0.0

closed 2023-05-07 by bernhard-42