bernhard-42/vscode-ocp-cad-viewer #8: Joints of Compounds are not shown

JonnyHaystack, 2023-04-30:
As the title says, although build123d allows you to add Joints to Compounds, they are not shown in the tree and model viewer

bernhard-42, 2023-05-07:
@JonnyHaystack Not sure I fully understand this. 

```python
b1 = Box(1, 2, 3)
b2 = Pos(1, 2, 3) * Box(1, 2, 1)

c = Compound.make_compound([b1 + b2])
c.label="c"
j = RigidJoint("j", c,  Plane(c.faces().sort_by()[-1]).to_location())
show(c, render_joints=True)
```

shows 
[image]

Do you have an example that doesn't work?

bernhard-42, 2023-06-19:
@JonnyHaystack Is there still something open?

JonnyHaystack, 2023-06-19:
Sorry haven't had a chance to try and reproduce this again

bernhard-42, 2023-12-12:
Will close now, please re-open, if it happens again

closed 2023-12-12 by bernhard-42