bernhard-42/vscode-ocp-cad-viewer #12: exception in config.py:status()

Thomas-Henkel, 2023-06-02:
vs-code Version 1.78.2 installed on windows10 Version 20H2
installed ocp_vscode and build123d

[image]

exception KeyError ["text"] when running a build123d example using show_object(), config.py:status() called by config.py:combined_config() while use_status=False, after requests.get(f"{CMD_URL}:{port}/status") returns {}. only calling status() when use_status=True worked around the problem for me.

bernhard-42, 2023-06-02:
I have problems in reproducing with the same versions of ocp_vscode and ocp_tessellate:

```python
from build123d import *
from ocp_vscode import *

b = Box(1,2,3)

show_object(b)
```
works

Is the cad viewer extension also 1.0.0
[image]

Thomas-Henkel, 2023-06-03:
Hi Bernhard,

yes, the viewer is v1.0.0 (see bottom of mail). The problem seems to be the
answer {} from port 3939 when status is queried. The query config seems to
work just fine. I tried it with the browser entering 127.0.0.1:3939/status
and also saw {} as the only answer. When I tried to reproduce this today,
the effect was gone. At the same time, when I saw this happen, I also
always got a completely black widget for the OCP viewer. Today, I could not
reproduce the issue, the  widget came up with the ocp logo when started and
the browser showed a status that contained a "text" key. I tried restarting
the viewer and vscode multiple times but could no longer reproduce the
problem.

I saw in a previous issue that the problem with the blank viewer at startup
happened to others, as well, with the effect disappearing after a while.
Very strange.

Now I just tried one last time restarting the viewer and there it is, again:

[image: image.png]

and here, what the browser says:

[image: image.png]
best regards,

Thomas
[image: image.png]

bernhard-42, 2023-06-03:
Thanks for the details, I will check when {} will be returned and where I need to add verifications that text is available.
btw., unfortunately, the images are not visible here on github

bernhard-42, 2023-06-04:
I still can't reproduce, but returning `{}` from the API call can only happen when 
1. either the splash scene doesn't appear, or 
2. when `status()` is called before the callback from showing the splash scene has been propagated.

I have changed `status()` to return `{}` when the API call returns `{}`
This should help with 2.

Case 1. is a different problem that I still haven't understood. The CORS topic shouldn't be the reason any more, since the extension does not access the Internet for getting the three-cad-viewer component any more, but gets everything from the extension local node_modules folder.

I will further research it ...

Thomas-Henkel, 2023-06-04:
Hi Bernhard,

In my case, returning {} clearly correlates with the missing splash scene.
If I have the blank widget, I can check at any time later with the browser
and still get {} for the status and the splash scene doesn't show up later,
either. Your fix for dealing with the {} return value will work fine for
me. The missing splash scene does not prevent me from showing an object
from my code, later, given I'm not getting the exception, first.

Thanks for looking into this.

Best regards,

Thomas

Thomas-Henkel, 2023-06-07:
[image]
here the extension host output for a vscode restart ending up with a blank ocp widget.

I then restarted ocp multiple times until I hit the blank widget, again. Nothing was added to the extension host output.

In the Developer Tools Console, I see

[image]

Multiple ocp restarts and one execution of box.py later, the Console looks like this

[image]

bernhard-42, 2023-06-07:
Great, thank you.
The line with `glass` is the issue. Now I know what to search for!

bernhard-42, 2024-01-16:
Closing for now. If it still is an issue, please re-open

closed 2024-01-16 by bernhard-42