bernhard-42/vscode-ocp-cad-viewer #17: `show_all` fails if you have a sketch that uses a solid's face as a workplane

shayded-exe, 2023-06-24:
It seems simply having a sketch that uses the face of a part as the workplane causes `show_all` to blow up.

Here's my code:
```py
# %%
from build123d import *
from ocp_vscode import *

# %%
reset_show()
show_clear()
set_defaults(
   axes0=True,
   helper_scale=50,
   reset_camera=False,
)

grid_unit = 42 * MM
baseplate_height = 7.2 * MM
baseplate_corner_radius = 4 * MM

with BuildPart() as baseplate:
    Box(
        length=grid_unit,
        width=grid_unit,
        height=baseplate_height,
        align=(Align.CENTER, Align.CENTER, Align.MIN),
    )

    top_face: Face = baseplate.faces().filter_by(Axis.Z).sort_by(Axis.Z)[-1]
    with BuildSketch(top_face) as plan:
        RectangleRounded(
            width=grid_unit,
            height=grid_unit,
            radius=baseplate_corner_radius,
            align=(Align.CENTER, Align.CENTER),
        )

show_all()
```

Which produces the error:
```
File c:\Users\ryans\miniforge3\envs\cad\Lib\site-packages\ocp_vscode\show.py:441, in show(**truncated**)
    438 progress = Progress([] if progress is None else [c for c in progress])
    440 with Timer(timeit, "", "overall"):
--> 441     data = _convert(
    442         *cad_objs,
    443         names=names,
...
    791         and coordSystem1.YDirection().IsEqual(coordSystem2.YDirection(), 1e-6)
    792         and coordSystem1.Direction().IsEqual(coordSystem2.Direction(), 1e-6)
    793     )

AttributeError: 'OCP.TopoDS.TopoDS_Face' object has no attribute 'Position'
```

If you change the sketch line to `with BuildSketch() as plan:`, (i.e. use the default XY workplane), it works fine.

shayded-exe, 2023-06-24:
I just discovered that wrapping the face in a `Plane` first works!

```py
with BuildSketch(Plane(face)) as plan:
```

It seems that `ocp_tesselate` is expecting workplanes to be actual planes, not faces here.
https://github.com/bernhard-42/ocp-tessellate/blob/0fd5aad8534932cd682818d382132c0af12aad9f/ocp_tessellate/convert.py#L639-L646

I'm not sure the ideal location for a fix. Either build123d could ensure faces get wrapped in planes sooner up the chain, or ocp_tesselate could check that it's actually dealing with a plane.

LeLocTai, 2023-06-25:
I'm just trying the example in the README and is having the same error. The example doesn't contain any BuildSketch.
```python
from build123d import *
from ocp_vscode import *

with BuildPart() as bp:
    with PolarLocations(3, 8) as locs:
        Box(1, 1, 1)

show_all()
```

bernhard-42, 2023-06-25:
@shayded-exe Agreed, this is a bug in ocp-tessellate. btw. `show(plan)` fails, `show(plan.sketch)` works.

bernhard-42, 2023-06-25:
@LeLocTai I tried this on a fresh Python interpreter and it works.
Since `show_all` tries to show all variables, could it be that you have some other variable in scope?

bernhard-42, 2023-06-25:
note to myself:
```python
File [~/Development/CAD/ocp-tessellate/ocp_tessellate/convert.py:644](https://file+.vscode-resource.vscode-cdn.net/Users/bernhard/Development/CAD/build123d/~/Development/CAD/ocp-tessellate/ocp_tessellate/convert.py:644), in _to_assembly(names, colors, alphas, render_mates, render_joints, helper_scale, default_color, show_parent, loc, mates, instances, progress, is_assembly, *cad_objs)
    [636]         else:
    [637]             pg.add(part)
    [639] elif (
    [640]     is_build123d(cad_obj)
    [641]     and hasattr(cad_obj, "sketch_local")
    [642]     and not (
    [643]         len(cad_obj.workplanes) == 1
--> [644]         and is_plane_xy(cad_obj.workplanes[0].wrapped)
    [645]     )
```

`is_plane_xy` needs to check for faces instead of planes and convert them before check

LeLocTai, 2023-06-26:
> @LeLocTai I tried this on a fresh Python interpreter and it works. Since `show_all` tries to show all variables, could it be that you have some other variable in scope?

That is probably it. Sorry for the confusion.

bernhard-42, 2023-07-17:
fix with 1.2.2

closed 2023-07-17 by bernhard-42