bernhard-42/vscode-ocp-cad-viewer #21: Sometimes viewer shows more than expected

Paul8043, 2023-07-10:
During the development of a model I run many similiar versions of the python code. I make intensive use of the tree-icons of the ocp-cad-viewer (clicking parts on & off). Eventually I reach a state in the ocp-cad-viewer where the same thing seems to be rendered twice, but at a slightly shifted position. Up-to-now I cannot reproduce this effect, but it happens with different models. If this state is reached, it remains stable, all further code-runs show the duplicated output. The effect goes away, if inside VS-Code the folder is closed and re-opened again (without restarting VS-Code). I do not have more information at the moment.
```
from build123d import *
from ocp_vscode import *

set_port(3939)

plane_xy = Plane.XY

show_all()
```
BAD: output is duplicated (happens somewhere during the debugging-cycle)
[image]
GOOD: as it is expected (seen initially or after re-open of the folder)
[image]
Any hints how to catch more information?

Paul8043, 2023-07-14:
In the meantime I have learnt that the Plane.XY and the 3D-Cross for the axes are rendered very similiar. The left cross belongs to Plane.XY und right cross to the axes. Plane.XY and Axes must be coincident all the time, but sometimes they are not!

1. start OCP-CAD-Viewer
2. run the above python program

[image]

Verify that `@0` and `Ortho` are turned ON.

1. `@0` turn OFF
2. `Ortho` turn OFF
3. `Grid xy` turn ON
4. `Ortho` turn ON
5. `Axes` turn ON

Then you will see the problem:

[image]

bernhard-42, 2023-07-14:
- The grid indicator and the plane symbol use the same pattern. It is very clear when you show more than the plane
- The plane symbol is a CAD object, as such it has a bounding box. `@0` ensures the axis indicator is show at global `(0,0,0)`, without `@0` it is shown at the center of the bounding box
- If you rotate the last example a bit and click in the tree on `plane_xy` (shows the bounding box) you can see the difference

[image]
[image]

As such, Plane.XY and Axes will not coincident all the time. They are only coincident if `@` is checked.

Hope that helps

bernhard-42, 2023-07-14:
Try also 
```python
b = Box(1,2,3, align=Align.MIN)
p = Plane(b.faces().sort_by(Axis.Z)[-1])
show_all()
```
[image]

Paul8043, 2023-07-14:
`Hope that helps`

Not really, I consider this behavior as very weird. Maybe we should clarify the abstract concept behind the view-buttons first. A simple tool-tip is not sufficient to explain the implemented functionality. Let me explain, why the understanding of these things is so difficult:
1. Functionality behind `Axes`:
Thought this would show the global 3D-cross, which is located all the time at (0,0,0). 
2. Plane.XY has has the same origin (0,0,0). So both, should be coincident all the time!
3. Axes, Plane.XY and Grid
All 3 objects have an infinite size. So, a bounding box does not make much sense. All 3 should be handled in the same way, either all 3 should be CAD-objects or none. If, for rendering, the infinite size is reduced to some finite amount, this should be no fundamental problem, in this case they may get a bounding box. But because the finite size is arbitrarily chosen, this provides not much information.
5. Functionality behind `@0`:
Up-to-now I could not find out the intented use of this button. This could make sense for single a singleCAD-object. But what would be the functionality, if several objects are drawn? 

The point is: This things are not intuitive, they deviate strongly form the normal understanding of the world. And it takes a lot of time to learn these things.

My proposal would be:

If somebody needs axes (bounding axes) inside a bounding box, then this functionality should get a separate button. The Axes (global axes) should control only the global 3D-cross. This is the needed anchor for all further orientations!

bernhard-42, 2023-07-14:
> Thought this would show the global 3D-cross, which is located all the time at (0,0,0).

It does when you do not press `@0`

Axes and grid are just that: Axes and grid. That is, they have an origin and three orthogonal axes plus units for the grid. No one had said that axes and grid always have to be at the global origin. 
Sometimes it is really helpful to see the coord axes located at the bounding box center. Especially since the grid follows. You can easily check symmetry in case of it not being at the global origin (0,0,0)

Another (common!) way to view it is to see the coord system at `(0,0,0)` as the global coord system. 
But there are infinite local coord systems (e.g. of working planes). And the viewer has one additional (local) coord system it shows which by definition has the origin located at the center of the bounding box. To chose between the global and local coord system the checkbox `@0` exists

But, again, no one forces you to uncheck `@0`. Then it would behave as you want.

> Plane.XY has has the same origin (0,0,0). So both, should be coincident all the time!

No one had said that axes are the global coord system, see above.

In a perfect world, the symbol for Plane.XY would not have a volume or a bounding box. But this is currently not supported. So the axes symbol for planes is a helper for debugging. If you don't use `show_all` and `show(plane)`, you will never see it. But for more complex objects it is helpful (and it is completely irrelevant whether it has some arbitrary bounding box, it does not impact your model)

And, they do unless you uncheck `@0`

> Axes, Plane.XY and Grid

This is a viewer for CAD objects. Hence axes and grid are chosen in size to be large enough for the radius of the bounding box of all shown objects. The world outside is irrelevant for the viewer. It is not a real world twin, it is a CAD object viewer.
So, whenever you show the grid, it is large enough for the whole object ensemble, e.g for every orthogonal view (front, top, left, ...)

> This things are not intuitive

I am sorry that it is not intuitive for you, but I haven't received a similar feedback in the 3 years this viewer exists.
And I hope my explanations help to understand it.

> they deviate strongly form the normal understanding of the world

I don't agree with this, see above

> The Axes (global axes) should control only the global 3D-cross

How about not unchecking `@0`? 
Some people like this functionality, but there is no need to use it

Paul8043, 2023-07-15:
I have tried to collect information from various channels. Primarily I start with README.md of the package, then I look for examples. From Discord I have only digested the last few weeks (not the last 3 years). 

Both Build123d and OCP-CAD-Viewer are great tools. I appreciate to have a viewer that can be run inside of VS-Code.

Thanks for your further explanations. And you are right, what is considered as intuitive is  subjective. It may vary from person to person.

OCP-CAD-Viewer is the most important tool for doing some work with Build123d. Therefore I need to understand completely what the view-buttons do.

Ok, now I know what is behind `Axes` and `@0`. We can tick them off.

Let us continue whith the remaining things:

- I could not find out what is the difference between the reset-button and the iso-view. Seems only the angle is slightly different. Is there more?
- Seems that OCP-CAD-Viewer tries to preserve a previously rendered model as long as possible. This makes sense in most cases. But sometimes you see an old version, either because of you have overlooked a syntax-error or because of all `show_object()-calls` have been uncommented! To make debugging easier I would prefer to start with an empty canvas. Seems that `show_clear()` provides this functionality. But there is also a `reset_show()` with an unclear behavior. When should `reset_show()` be used?
- Debugging
`"during debugging, show and show_object are disabled. They interfere with the visual debugging"`
What about the others like `debug()` and `show_all()`? Can you sketch roughly how debugging works internally? This is just for interest and would help to establish the right concept in my brain ! 
- Quickstart build123d
This works only, if an editor is open. This is  easily forgotten. Is there a chance to get rid of this restriction? Would it be possible to store this information in the viewer preferences? With the hope that this needs to be done only once. The `Quickstart` takes long and it seems that the startup-times vary strongly. I have started VS-Code with the correct mamba environment, but I see a lot of command-windows popping up which try to activate the same environment again, by using `conda` instead of `mamba.` Could not find out from where this behavior comes from. Could be that this has nothing to do with OCP-CAD-Viewer. But this generates very long startup-times (sometimes up to 20 secs).

bernhard-42, 2023-07-15:
> I could not find out what is the difference between the reset-button and the iso-view. Seems only the angle is slightly different. Is there more?

Try to use either zoom (iso-view does not reset the zoom) or `reset_camera=Camera.KEEP` or `reset_camera.CENTER`. Now the viewer will preserve position and rotation of the camera (and with KEEP also the panning position).
Then iso-view switches to iso-view and reset-button to the memorized camera position

>  When should reset_show() be used?

`show_objects` shows the objects incrementally (like in CQ-Editor). To reset the list of objects (in Python) to be shown, use `reset_show`. The next object shown with `show_object` will be the only one shown.

`clear_show` simple clears the viewer. It doesn't touch the object list in Python as related to `show_object`. It is mainly used for visual debugging.

> Debugging

I have launched a new version today, v 1.2.2. Please look at the change log, this is now more intuitive.

I have a listener in the VS Code extension that gets called when one executes a step in the debugger. This listener then calls `show_all` in Python. In the new version, this `show_all` is suppressed when the last command in the debugger was a `show` or `show_object` call.
The interworking of Python, VS Code extension and Javascript viewer component is pretty complex, but the above summarizes the approach

> Quickstart build123d
This is a nasty restriction where I have no idea how to overcome it.
The viewer needs to correct Python interpreter path. But this path is only available when a Python file is open.
I don't really like the "No editor open" error, but have no idea currently how to avoid it.

Quickstart is a complex activity: It completely sets up the Python environment.
If you install build123d in your environment, then it will go much faster.
I usually use quickstart only once for my CAD Python environment. And then I use this environment for all future projects. In this case the quickstart doesn't even appear.

Paul8043, 2023-07-17:
I have made good progress during the last week end. I have to revise my first assessment related to bounding-boxes. They are really helpful. Sorry for being somewhat impatient. I would like to provide you with some more feedback. Could you activate the Discussions-Tab of this repository? I think this would be the better place for feedback.

bernhard-42, 2023-07-18:
opened the discussions

closed 2023-12-09 by bernhard-42