bernhard-42/vscode-ocp-cad-viewer #28: OCP Image Import

jdegenstein, 2023-07-19:
Like many other CAD software it would be great to import a raster image into OCP CAD Viewer. That way you could take a photo of a real-life object, import/rescale in OCP CAD Viewer, and then design/sketch on top of the image in build123d/CQ. Flatbed scanners can be really good sources of reference images as well.

Desired features:
- Import a variety of raster image formats (PNG/JPG/etc)
- Have the ability to scale the image uniformly
- Have the ability to adjust transparency of this image-plane
- Have the ability to place the image-plane on an arbitrarily defined Plane with appropriate defaults like Plane.XY, Plane.XZ etc

As far as ThreeJS is concerned, here is a working example someone created of loading an image onto a plane as a texture:
https://codepen.io/duhaime/pen/jaYdLg

I suggest looking at the Python Pillow library (there may be lighter weight alternatives). This may be relevant for encoding an image to an IO object in memory.
https://pillow.readthedocs.io/en/latest/reference/Image.html#PIL.Image.Image.tobytes

This SO question may also be relevant:
https://stackoverflow.com/questions/27826422/get-binary-image-data-from-pil-image

Yeicor, 2024-01-13:
I was about to suggest this too, but I would generalize to import textured meshes to use as a reference while designing. This would be interesting for loading 3D scans or other models without losing texture information.

There could be a helper function for loading an image that would create the model described above and load it as a textured mesh.

bernhard-42, 2024-03-23:
fixed in 2.2.2

closed 2024-03-23 by bernhard-42