bernhard-42/vscode-ocp-cad-viewer #33: Feature: Support naming nested objects in show()

philipsd6, 2023-10-11:
It's possible to send multiple objects to `show(obj1, obj2)` but what's really nice is that you can send nested objects as well, i.e.:
`show(main_object, parts_list)` and they show up hierarchically in the OCP Cad Viewer. In this case the top level objects can be named as documented:
```python
show(main_object, parts_list, names=["Main", "Parts"])
```
but it doesn't seem possible to name the nested objects in the `parts_list`. I would like to request that this be supported, perhaps with syntax like:
```python
show(main_object, parts_list, names=["Main", {"Parts": ["Widget A", "Widget B", ...]})
```

bernhard-42, 2023-10-11:
Try
```python
b = Box(1, 1, 1)
s = [Pos(2 * i, 0, 0) * Sphere(0.2)  for i in range(10)]

for i in range(10):
    s[i].label = f"sphere_{i}"

show(b,s, names=["b", "s"])
```
[image]

philipsd6, 2023-10-13:
Now I'm just making sure to label all the objects and don't need to use the `name` parameter at all!

closed 2023-10-13 by philipsd6