bernhard-42/vscode-ocp-cad-viewer #35: Request: Auto-restart open viewers when launching VSCode

MatthiasJ1, 2023-11-03:
If there is a viewer tab already open when VSCode is launched, you need to manually go to the OCP Cad Viewer menu and restart the viewer before it works.

bernhard-42, 2023-11-24:
A faster way is pressing `cmd-k v` on Mac or `ctrl-k v` on windows

bernhard-42, 2024-01-17:
@MatthiasJ1 Now that OCP CAD Viewer starts the viewer e.g. on save, and typically on opening a Python file, is this still open?

zwn, 2024-02-03:
For me it does not work. I get messages like

[image]

even though I have workspace and editors open. When I try to open the viewer, I get a message stating that I should first open a file in my project. I need to first click into the editor area and only then I can try starting the viewer. It tries to start but tells me

```
Viewer backend started
[Errno 111] Connection refused
```

and quits. Then I go check my terminals and I see several `OCP backend` open, one of them with existing backend. Only when I go and close the viewer window (which is black, not showing anything), I can open the viewer and get it working. I also get some editor doublings and windows reshuffling when I close the IDE and start it again.

My setup involves 3 python files open in tabs, each split into two cells (as suggested in readme), each attached to its own kernel, and one viewer window. Once I get it started, it works like a charm but getting there takes some effort. I am on debian 12.

bernhard-42, 2024-02-03:
Let's check a scenario that works:
- use a terminal
- activate your Python environment, e.g. `conda activate myenv`
- change to folder where your Python files are
- start VSCode with `code .`
- open a Python file from the file explorer in VSCode
Is it working then?

zwn, 2024-02-03:
I can confirm this scenario works for me.

bernhard-42, 2024-02-10:
Closing for now. Please reopen if you feel that it does't work as expected

closed 2024-02-10 by bernhard-42