bernhard-42/vscode-ocp-cad-viewer #36: doesn't work anymore

incredible-io, 2023-11-15:
Hello.

I had setup cadquery and the viewer two weeks ago and it worked. But today, after two weeks pause, it doesn't show the ocp logo, when I open the viewer, and also my models.

The log doesn't show any error:
```
[17:17:49.710} INFO  ] Successfully registered CadqueryViewer Status Manager
[17:17:49.711} INFO  ] Successfully registered CadqueryViewer Library Manager
[17:17:50.183} DEBUG ] Reviving webview panel
[17:17:54.319} INFO  ] ocp_vscode library version 1.2.2 matches extension version 1.2.2
[17:17:54.319} INFO  ] Server started on port 3939
[17:17:54.319} INFO  ] Starting websocket server ...
[17:17:54.319} DEBUG ] Revealing existing webview panel
[17:17:54.320} DEBUG ] Updateing webview
[17:17:54.320} DEBUG ] Command cadqueryViewer registered
```

I had updated vscode since last successful run and today jupyterlab, but nothing more changed.
Do you have any suggestions?

Regards,
Cid

bernhard-42, 2023-11-15:
Could you open the developer tools to see the brwpser javascript console?
Is there any error visible

incredible-io, 2023-11-16:
Hello.

Thank you for the hint. 👍🏼 The following error is repeatedly printed to the console:

```
WebGL: INVALID_OPERATION: drawElementsInstanced: no valid shader program in use
three-cad-viewer.esm.js:19914 THREE.WebGLProgram: Shader Error 1282 - VALIDATE_STATUS false

Program Info Log: Program binary could not be loaded. Binary is not compatible with current
driver/hardware combination. Driver build date Oct 26 2023. Please check build information of
source that generated the binary.
```

This happened with current graphics driver `nvidia-open-dkms 545.29.02`. After downgraded to previous version `535.113.01` the error went away and it start working again.
Can you fix the `three-cad-viewer.esm.js` or update dependencies, maybe it is fixed already, to make it compatible with recent gpu driver?

Regards,
Cid

bernhard-42, 2023-12-09:
The only thing I can do is add the newest threejs version ...
Next version will have the latest, but no idea whether it'll fix it

bernhard-42, 2023-12-12:
Is 2.0.13 working with the latest nvidia-open-dkms driver?

bernhard-42, 2024-01-17:
Since I can't do anything more, closing it

closed 2024-01-17 by bernhard-42

incredible-io, 2024-01-21:
Hello.

Sorry, I had other priorities last weeks and not thought of it.

Now I have updated the extension with all dependencies to v2.1.1 and it is still incompatible to the latest driver. Sad, but it is not so important for me in the moment and I will try it again with the next updates.

Thank you for your efforts.

Regards,
Cid