bernhard-42/vscode-ocp-cad-viewer #42: KeyError: 'measure_tools'

sstiene, 2023-12-09:
I have just started using ocp viewer. I managed to install it and execute the viewer.

However starting a minimal example results in KeyError: 'measure_tools'

Minimal example:

import cadquery as cq
from build123d import *
from ocp_vscode import *
b = Box(1, 2, 3)
b = fillet(b.edges(), 0.2)
c = Pos(2, 2, 0) * Cylinder(0.5, 2)
#
# Use config with show for one time setting configurations
#
# %%
# Use default settings
show(b, c)

Error:

Traceback (most recent call last):
  File "cadquery_test.py", line 14, in <module>
    show(b, c)
  File "C:\Users\stefa\AppData\Local\Programs\Python\Python310\lib\site-packages\ocp_vscode\show.py", line 441, in show
    measure_tools = conf["measure_tools"]
KeyError: 'measure_tools'

bernhard-42, 2023-12-09:
Looks like you are using the extension from the Marketplace (1.x) and the latest version of ocp-vscode (2.x9
We are currently in a beta test phase that requires ocp-vscode 2 to be available.
Either install ocp-vscode from within the extension in vs code (Library manager) that installs ocp-vscode 1.x or load and install the vsix from github.
I will try to finally launch 2.x in the VS Code marketplace this weekend

sstiene, 2023-12-09:
[image]

These are my versions. As you said: Its a little bit strange as ocp_vscode 2.0.6 is installed but ocp_vscode 1.2.2 is running... I am not aware that I installed something by hand. Just clicked on the "install ... buttons"

sstiene, 2023-12-09:
downgraded ocp_vscode to 1.2.2 and it works. Thanks

bernhard-42, 2023-12-09:
> I am not aware that I installed something by hand. Just clicked on the "install ... buttons"
hmm, interesting, I then need to check the commands what went wrong.
But glad it is working again

bernhard-42, 2023-12-09:
Indeed, for quick starts I did not check the ocp_vscode version.
Fixed it in the release, coming soon

bernhard-42, 2023-12-11:
2.0.13 is on VS Code Marketplace. So this shouldn't be a problem any more

closed 2023-12-11 by bernhard-42