bernhard-42/vscode-ocp-cad-viewer #46: Latest version always requires `build123d`

jeremiahrose, 2023-12-20:
In previoud versions, it was possible to use the `ocp_vscode` with CadQuery without having `build123d` installed. This was important for Apple Silicon users because there is currently no easy way to get both CadQuery and build123 installed in the same environment. Regardless of this, cadquery users shouldn't be required to install `build123d` for the extension to work.

bernhard-42, 2023-12-20:
We needed some higher level API to implement the backend for measurement.
Please see here how to install both cadquery and build123d: https://build123d.readthedocs.io/en/latest/installation.html#special-notes-on-apple-silicon-installs

jeremiahrose, 2023-12-20:
If `build123d` is actually required, then perhaps we should add it to the package dependencies?

bernhard-42, 2023-12-21:
agreed, we've missed it.
I am already working on removing the build123d dependency, so this shouldn't be a problem in the next release any more

bernhard-42, 2024-01-13:
Fixed in latest version

closed 2024-01-13 by bernhard-42