bernhard-42/vscode-ocp-cad-viewer #50: Can be added to the install instructions for Arm Macos

lishine, 2024-01-04:
brew install miniforge

mamba init "$(basename "${SHELL}")"
mamba create -n code_cad python=3.11
mamba activate code_cad
code .

Install python extension
Install OCP CAD Viewer extension

bernhard-42, 2024-01-16:
Added it to the Readme

lishine, 2024-01-23:
@bernhard-42 
I saw you added the documentation, but it still confuses me.
When I first got here, I was overwhelmed. But the thing is that it is very very simple.

That confuses me:

```
Necessary tools:
python and pip available in the Python environment that will be used for CAD development
```
As if I need to install pipe or python, I don't.

And the arm instructions, it says go back to #1. It confuses. 

As a user I want start the page with 3 titles - windows/Linux/mac/mac-arm.
Mac and mac arm can have the list of instructions that I wrote above and it will be very simple , otherwise it puts off. 

BTW I think this is important:

```
add to the end of your shell profile ~/.zshrc
conda activate code_cad
```

Otherwise env not activated

bernhard-42, 2024-01-25:
well, without python and pip nothing works. And they are not installed on every platform by default ...

Unfortunately your instructions are not fully consistent. It doesn't make sense to use a command to identify the shell and then later add something to .zshrc (I for example do not use zsh but bash).

And you definitely do not want to add the activation of your env to your zshrc. Most people work with many envs and activate the one they need ad hoc.

Hence I think my description is a bit more consistent and much more generic.
If you only work with one env, do what you described. But we can't add that to the Readme.

closed 2024-01-25 by bernhard-42

lishine, 2024-01-26:
>well, without python and pip nothing works.
Of course not. Python and pip work from the venv. No need the global ones.
By default shell not activates venv and then it will use the global ones instead of erroring , which is what I suggest preventing by not installing global ones. It is not essential and moreover confusing.
I am telling you that I am as a typescript programmer coming to this extension , looked at the instructions, was overwhelmed and thought I would have to compile from source or something...
As a Mac and typescript user , I want to see easy few lines of straight instructions. I mean that will ease bringing users.

bernhard-42, 2024-01-26:
I got asked by users to add this statement, because they e.g. didn't have pip installed. They were confused. Hence I added it.
So, what is confusing or not depends on the user :-)

lishine, 2024-01-26:
😅