bernhard-42/vscode-ocp-cad-viewer #54: Model is not visible

silverlore, 2024-01-31:
When I Open the viewer I can not see the Eye model. I can not even see the XY grid.
If a select one of the sub models f.eks. Eye then the selection box become visible but nothing else
[image]

[image]

[image]

If I switch to the clipping tab then the model is visible.

silverlore, 2024-02-02:
It works now. might have been a vscode version problem

closed 2024-02-02 by silverlore

Jest0r, 2024-02-20:
I have the same problem since today (ubuntu 22.04).  No objects are visible, neither the original Eye or any own object.
VSCode 1.86.2

bernhard-42, 2024-02-20:
Do you see any errors in the output window of VS Code or of the Extension host?
And if you toggle on the developer mode (javascript console) any errors there?

Jest0r, 2024-02-22:
No errors that I could spot, and the problem persisted over multiple reboots.
I also changed and checked the port, all seemed fine.

When I wanted to copy/paste the outputs today, the problem disappeared. All back to normal.

Jest0r, 2024-02-26:
it seems that it works on and off, but I can't determine when. Currently it fails most of the time.
The js console indeed throws a lot of errors - see here: 
[image]

bernhard-42, 2024-02-26:
Let's check a scenario that works:

- use a terminal
- activate your Python environment, e.g. `conda activate myenv`
- change to folder where your Python files are
- start VSCode with `code .`
- open a Python file from the file explorer in VSCode

Is it working then?