bernhard-42/vscode-ocp-cad-viewer #55: Picking for measurement tools or "show shape properties" does not honor clipping planes

MatthiasJ1, 2024-02-08:
Occlude a feature with a shape and then hide that shape using the Clipping tab. Attempting to select the feature will only select the occluding shape.

bernhard-42, 2024-02-09:
Clipping isn't meant to fully hide objects. Please use the eye/mesh buttons for that.
The two views "Tree" and "Clipping" solve different tasks

MatthiasJ1, 2024-02-09:
In my scenario, I have a single object which has interior features. I wanted to use the measuring tools to inspect an internal face so naturally I used the clipping planes feature to expose the interior. When I attempt to select the clearly visible interior feature, it selects the non-visible exterior feature.

bernhard-42, 2024-02-10:
Understood, but afaik, this is not how threejs works. Clipping and hiding are different concepts. As far as I understand, clipping tells the renderer which part of the object  to render, hiding will remove it temporarily from the scene. Selecting works with raycasting and the rays bypass hidden objects, but not clipped objects.
I am not sure I really understand these aspects, however, I use threejs mostly as is and don't / can't change its behaviour.

Alternatively, show object.faces() instead of object for the surrounding object, then you can remove the faces that are in the way

bernhard-42, 2024-09-03:
Will close now, since clipping is disabled in measure mode anyhow

closed 2024-09-03 by bernhard-42