bernhard-42/vscode-ocp-cad-viewer #58: Request: Measure tools toggle in GUI

MatthiasJ1, 2024-02-09:
Since the measuring tool kit is something that is used purely with mouse interaction in the GUI window, it would make sense to have the controls related to enabling it in the GUI rather than as `measure_tools=True` in the code.

bernhard-42, 2024-02-10:
I decided against in the first version of this feature, since this would have made an anyhow complex feature even more complicated. 
Implementing this feature was difficult enough and I guess I won't find the motivation  to add new behavior to the javascript part of this feature anytime soon

sgtkt, 2024-05-04:
Although currently we cannot toggle the measurement tools in the GUI, we can enable it by default in the settings.

[image]

bernhard-42, 2024-09-03:
added in 2.4.0

closed 2024-09-03 by bernhard-42