bernhard-42/vscode-ocp-cad-viewer #59: Add a parameter to suppress show sketch_local

bernhard-42, 2024-02-13:
Jern's approach:

```python
classes = (BuildPart, BuildSketch, BuildLine)  # for OCP-vscode
variables, s_o, s_n, slocal = (list(cf().f_locals.items()), [], [], False)
for name, obj in variables:
    if (
        isinstance(obj, classes)
        and not name.startswith("_")
        and not name.startswith("obj")
    ):
        if obj._obj_name != "sketch" or slocal:
            s_o.append(obj), s_n.append(f"{name}.{obj._obj_name}")
        elif obj._obj_name == "sketch":
            s_o.append(obj.sketch), s_n.append(f"{name}.{obj._obj_name}")
show(
    *s_o,
    # ,
    names=s_n,
    reset_camera=Camera.KEEP,
)
```

jdegenstein, 2024-02-13:
missing import statement for the above:
```py
from inspect import currentframe as cf
```

bernhard-42, 2024-03-23:
fixed in 2.2.2

closed 2024-03-23 by bernhard-42