bernhard-42/vscode-ocp-cad-viewer #60: Add a 1-3 axis plane view

CePeU, 2024-02-23:
OpenScad and many CAD programs offer the possibility to see a XY, ZY,, ZX plane on which (or in relation to which) your object is placed. Also many times a scale is shown on those axis (ideally you can choose a scaling). It would be helpfull if one could also toggle which planes are shown, toggle scaling, input scaling size and thus to see directly how ones build object is located on the X,Y,Z planes.

bernhard-42, 2024-02-23:
there is the ability to turn on grid planes, however they are at the outside of a bounding box. You can select which grid plane(s) to show.
In the info box (under the navigation tree), the scale of the grid is given.
You can control grid by providing the `ticks` parameter (note, the algorithm adapts ticks to have nice values)

bernhard-42, 2024-02-23:
And if you use top, bottom, right, left, back, front view in ortho projection, the objects looks like being on the grid  plane

CePeU, 2024-02-24:
Ok I was not aware what ticks stands for nor did I find it in the description as valid parameter (but I might have overlooked it). I will play around with tick.
The ability to show a grid plane I was aware of but when using 3D view it allways is problematic because of your viewpoint and skewing. I also used the ortho projections but they feel more like a workaround (although they DO have their purpose and they are helpfull).

I still would love to have something like this
[image]

bernhard-42, 2024-02-27:
The show parameters are described here: https://github.com/bernhard-42/vscode-ocp-cad-viewer/blob/main/docs/show.md

It is a tricky problem. Assume your cube is not at the origin, then you have the same issue. I onky see two options: Center the grid at (0,0,0) or at the outside of the world bounding box (as today). None covers all cases well

bernhard-42, 2024-03-23:
fixed in 2.2.2

closed 2024-03-23 by bernhard-42