bernhard-42/vscode-ocp-cad-viewer #61: Settings parameter KEEP_PANEL_OPEN to not close the terminal

42sol-eu, 2024-02-26:
In some use-cases it is not the intended behaviour of the user to let OCP-View close the terminal panel.

Expected behaviour: 
- add a settings parameter (bool) to prevent OCP-viewer from closing the terminal panel after startup.

if new parameter `KEEP_PANEL_OPEN` =false : OCP-viewer will close the output/terminal panel
if new parameter `KEEP_PANEL_OPEN` =true : OCP-viewer will not touch the sate of the output/terminal panel

42sol-eu, 2024-02-26:
> My use case is to call python scripts from the terminal.
> the closing of the terminal is therefore unintended, though only a tiny anoyance

Thanks for your great work on the viewer itself. I really value your time and effort.
If you are ever near Kassel - you can contact me for a small tour and a free lunch

bernhard-42, 2024-02-27:
Had already planned to make it configurable (since I also don't like the bevavior any more) It is useful for people with smaller screens, hence I will introduce a VS Code workspace config (no need to use set_defaults)

bernhard-42, 2024-02-27:
Glad you like it and thanks for the offer😀

bernhard-42, 2024-03-23:
fixed in 2.2.2

closed 2024-03-23 by bernhard-42