bernhard-42/vscode-ocp-cad-viewer #68: Add new CQ wheels so that Apple Silicon users don't need the mamba/miniforge workaround

rawwerks, 2024-03-21:
@jdegenstein has kindly addressed the upstream issue with CQ not having build wheels for Apple Silicon. 

Would you be willing to update this plugin to take advantage of the new build wheels, so that us Apple Silicon users don't need the workaround? 

(FWIW - I personally have not been able to use this extension because the mamba workaround doesn't work for me.)

jdegenstein, 2024-03-21:
For clarity, the relevant release is here: https://github.com/CadQuery/ocp-build-system/releases/tag/7.7.2.0

I believe the command to use is e.g. `pip install https://github.com/CadQuery/ocp-build-system/releases/download/7.7.2.0/cadquery_ocp-7.7.2-cp311-cp311-macosx_11_0_arm64.whl` obviously needs to be adjusted to provide the best pip method along with the correct python version.

The CQ folks have requested a storage increase with PyPI but it hasn't received a response yet here https://github.com/pypi/support/issues/3631

@rawwerks if you want to use the extension you should be able to simply pip install cadquery-ocp yourself

infused-kim, 2024-05-25:
It looks like the storage increase has been approved. 

Is there anything else standing in the way of getting cadquery-ocp for Apple silicon onto pip?

I would love to finally drop conda.

bernhard-42, 2025-03-02:
Fixed with 2.6.2

closed 2025-03-02 by bernhard-42