bernhard-42/vscode-ocp-cad-viewer #71: Regression in 2.2.2

MatthiasJ1, 2024-03-24:
Alignment tuples cause `show_all()` to error out.
```py
from build123d import *
from ocp_vscode import *

MIDDLE = (Align.CENTER, Align.CENTER)
show_all()
```
```
File "ocp_vscode/show.py", line 815, in ocp_group
    return group.objects[0]
           ~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

bernhard-42, 2024-03-24:
yep, indeed.
The only thing that can be done at the moment until the fix is out is to exclude the variable 
`show_all(exclude=["MIDDLE"])`

bernhard-42, 2024-03-24:
This will be the replacement for `ocp_group` in `ocp_vscode.show.py`:

```python
def ocp_group(obj, name):

    def to_group(obj, name, group):
        if isinstance(obj, list):
            sub_group = OCP_PartGroup([], name=name)
            for i, el in enumerate(obj):
                to_group(el, f"{name}[{i}]", sub_group)
            if len(sub_group.objects) > 0:
                group.add(sub_group)
        elif isinstance(obj, dict):
            sub_group = OCP_PartGroup([], name=name)
            for name, el in obj.items():
                to_group(el, name, sub_group)
            if len(sub_group.objects) > 0:
                group.add(sub_group)
        else:
            if (
                (is_wrapped(obj) and not obj.__class__.__name__ == "Color")
                or is_build123d(obj)
                or is_cadquery(obj)
            ):
                new_obj = conv(obj)
                new_obj.name = name
                group.add(new_obj)
            else:
                pass
                # print(
                #     f"show_all: Type {type(obj)} for name {name} cannot be visualized"
                # )

    group = OCP_PartGroup([], name=name)
    to_group(obj, name, group)
    return group.objects[0] if len(group.objects) >= 1 else group
    ```
If you would find some time to give it try against all sort of variables that come to your mind, this would be really help to harden `show_all` :-)

MatthiasJ1, 2024-03-27:
Heres some additional bugs (although I haven't tested your replacement yet)

---

```py
a = [Pos(1,1)]
show_all()
```
```
RuntimeError: Cannot transform [<OCP.TopLoc.TopLoc_Location object at 0x14f956a70>], e.g. mixed Compounds not supported here?
```
---
```py
a = Pos(1,1) # or Vector(1,1)
show_all()
```
[image]

bernhard-42, 2024-03-27:
Great, thanks, I'll test it (and fix if needed)

MatthiasJ1, 2024-03-28:
Another 2.2.2 regression: Lists of objects no longer display their color properly.

```py
p = Sphere(1)
p.color = Color(0x00FF00)
```
[image]

```py
p = [p]
```
[image]

Prior to the latest release this would still display green. This is needed in scenarios such as coloring each face of a shell based on their properties.

bernhard-42, 2024-03-28:
Thanks for raising these issues. I am away from keyboard until mid of next week and will look into it then

MatthiasJ1, 2024-03-30:
Tuples also no longer work
```py
p = [Circle(1), ]
show_all() # ok

p = (Circle(1), )
show_all() # <-- ERROR
```

MatthiasJ1, 2024-03-30:
This fails with the new `ocp_group`
```py
loc = GridLocations(1, 1, 5, 5)
show_all()
```
```
RuntimeError: Cannot transform [<OCP.TopLoc.TopLoc_Location], e.g. mixed Compounds not supported here?
```

bernhard-42, 2024-03-30:
`ocp_group` was meant to be a shortcut. Seems like there are no really shortcuts in life. I need to properly integrate with ocp_tessellate. Will do next week.

MatthiasJ1, 2024-03-31:
Another test case when you get around to looking at it. This will crash vscode:
```py
s = Circle(1) - Circle(1)
show_all()
```

MatthiasJ1, 2024-04-15:
Coloring objects within a list no longer works. Before 2.2.2 this would display the shell with each face colored based on its `geom_type` whereas now it displays everything as the default purple face color:

```py
faces = []
for f in s.faces():
        f.color = Color(colors[f.geom_type])
        faces.append(f)
```

bernhard-42, 2024-04-17:
Quick update: I implemented unit tests for `show` and `show_all`. Your test cases are covered: https://github.com/bernhard-42/vscode-ocp-cad-viewer/blob/aea58df2b1b8bdca663bde36011342c4e5b3159a/pytests/test_show.py#L195-L336

bernhard-42, 2024-05-01:
Fixed in the last version

closed 2024-05-01 by bernhard-42

MatthiasJ1, 2024-05-08:
Been busy recently and just got around to testing these, everything seems to be fixed! Thank you!