bernhard-42/vscode-ocp-cad-viewer #72: Model is invisible in tree view

PaulBone, 2024-04-22:
My model, and even the OCP logo that it opens with.  Almost always are invisible in the OCP viewer window.  But many controls including the tree are visible.  If I click "Clipping" then I can see my model.  And sometimes it works fine but I don't know what makes it work or break.  I'm using Linux, otherwise I'm on the latest version of VSCode and the plugin.

[image]

Thanks.

bernhard-42, 2024-04-22:
Can you open the developer tools in VSCode (ctrl-shift-p toggle dev) and see wether you have opengl issues ?

PaulBone, 2024-04-23:
It's working today.

OCP vscode has updated since yesterday, it was 2.2.2 now it's 2.3.1.  And also I shut down the PC overnight.  I'm not aware of any other changes but over the last few weeks I found sometimes it worked and sometimes it didn't.  I've captured the logs from the developer console when it works.  I'll see if it fails again and capture logs then.

I'd like to keep the bug open in case the problem comes back and I can get more information.  Thanks.

PaulBone, 2024-04-24:
@bernhard-42 

When it doesn't work I see a bunch of errors like:

```
three-cad-viewer.esm.js:20276 THREE.WebGLProgram: Shader Error 0 - VALIDATE_STATUS false

Material Name: 
Material Type: LineBasicMaterial

Program Info Log: Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 19 2024. Please check build information of source that generated the binary.
Location of variable pc_fragColor conflicts with another variable.

onFirstUse @ three-cad-viewer.esm.js:20276
Show 1 more frame
Show less
60WebGL: INVALID_OPERATION: useProgram: program not valid
three-cad-viewer.esm.js:15634 WebGL: INVALID_OPERATION: drawArrays: no valid shader program in use
render @ three-cad-viewer.esm.js:15634
Show 1 more frame
Show less
three-cad-viewer.esm.js:20276 THREE.WebGLProgram: Shader Error 1282 - VALIDATE_STATUS false

Material Name: 
Material Type: LineBasicMaterial

Program Info Log: Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 19 2024. Please check build information of source that generated the binary.
Location of variable pc_fragColor conflicts with another variable.

onFirstUse @ three-cad-viewer.esm.js:20276
Show 1 more frame
Show less
123WebGL: INVALID_OPERATION: drawElements: no valid shader program in use
2three-cad-viewer.esm.js:20276 THREE.WebGLProgram: Shader Error 1282 - VALIDATE_STATUS false

Material Name: 
Material Type: MeshBasicMaterial

Program Info Log: Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 19 2024. Please check build information of source that generated the binary.
Location of variable pc_fragColor conflicts with another variable.

onFirstUse @ three-cad-viewer.esm.js:20276
Show 1 more frame
Show less
three-cad-viewer.esm.js:20276 THREE.WebGLProgram: Shader Error 1282 - VALIDATE_STATUS false

Material Name: frontMaterial
Material Type: MeshStandardMaterial

Program Info Log: Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 19 2024. Please check build information of source that generated the binary.
Location of variable pc_fragColor conflicts with another variable.
```

Today the first time VS code opened it opened my project but the OCP Viewer window was a blank dark grey (the background colour).  I closed it and restarted things and it went to not painting my model.

PaulBone, 2024-04-28:
I've tried a few different versions of my NVidia drivers.  Both older and newer.  Nothing worked.  (When you say "driver" I'm guessing that's what you meant?).

If I open https://bernhard-42.github.io/three-cad-viewer/example.html in my browser (Either Firefox or Chromium) it works fine.  So I guess it's something to do with VSCode's environment or the VSCode extension?

Is three.js something the extension packages or something that vscode provides?  I'm not familiar with JS packaging things, it looks like it might be something the viewer brings with it.

When the error message says:

> Program Info Log: Program binary could not be loaded. Binary is not compatible with current driver/hardware combination. Driver build date Mar 12 2024. Please check build information of source that generated the binary.
> Location of variable pc_fragColor conflicts with another variable.
> 

What binary is it talking about?  The shader? is it precompiled and cached somewhere?  or a program binary or shared library.

Thanks?

bernhard-42, 2024-04-28:
threejs is packaged by my three-cad-viewer which in turn is used by OCP CAD Viewer in VS Code.

bernhard-42, 2024-04-28:
I am not familiar with the internals of threejs which sits on top of WebGL in the "browser" environment of use. 
When you use e.g. Firefox to access the three-cad-viewer page, it uses WebGL as compiled into Firefox
VS Code provides so called webviews: they act like a browser in a sandbox in VS Code. So I would expect, WebGL comes from VS Code (which I think uses Electron)

bernhard-42, 2024-04-28:
I googled the binary shader thing, but couldn't make sense of what I found :(

Which Linux are you using? I read that Arch Linux for example does sometimes have issues

bernhard-42, 2024-04-28:
and yes, I meant the Nvidia drivers. Somebody got it working again with an older driver in the past. I think the shader stuff is a very low level stuff in webgl, way deeper in the stack as to where I use it (above threejs)

bernhard-42, 2024-04-28:
you said "when I use clipping then I can see my model".
Is this reproducibly always the case?

bernhard-42, 2024-04-28:
just for reference: https://github.com/electron/electron/issues/40684

bernhard-42, 2024-04-28:
if it is a cache issue  it would explain why it sometimes works and sometimes not.
another reference https://bobbyhadz.com/blog/vscode-clear-cache#clearing-the-visual-studio-code-cache-on-linux

bernhard-42, 2024-04-28:
sorry for flooding you with a bunch of "blind shots", but I have no env to reproduce ...

PaulBone, 2024-04-29:
> I googled the binary shader thing, but couldn't make sense of what I found :(
> 
> Which Linux are you using? I read that Arch Linux for example does sometimes have issues

I'm using Mint, which is based on Ubuntu.  But for unrelated reasons I want to switch to Debian anyway.  Actually I have a debian PC I can try this on, and if that works it gives me hope that a reinstall may help.  Just it doesn't help us understand why it's not working or how to help if someone else has the same problem.

PaulBone, 2024-04-29:
> and yes, I meant the Nvidia drivers. Somebody got it working again with an older driver in the past. I think the shader stuff is a very low level stuff in webgl, way deeper in the stack as to where I use it (above threejs)

If it helps I know what shaders are.  I've done a little bit of OpenGL programming (most of it a long time ago before shaders existed ;-)) and a tiny tiny bit of WebGL.

PaulBone, 2024-04-29:
> you said "when I use clipping then I can see my model". Is this reproducibly always the case?

It was for a while, but lately it hasn't worked at all.  And I don't know if I had different errors then or not.

PaulBone, 2024-04-29:
> sorry for flooding you with a bunch of "blind shots", but I have no env to reproduce ...

No worries.  I really like build123d and I'm invested in getting it working.  I want a free CAD program suitable for a hobbyist that won't change its licensing terms in the future.  I tried FreeCAD too and it was okay but so far I've been more effective in build123d completing most drawings sooner there.  A little debugging/troubleshooting won't hurt me and can help me and other people.

bernhard-42, 2024-04-29:
I checked the cache folder of vs code on my Mac and there are actually quite a lot of cache files - binary, however containing javascript and similar parts.
That the clipping trick doesn't work any more, could be another indication of a cache issue.
So I'd say cleaning the cache could be worth doing and shouldn't destroy anything

PaulBone, 2024-04-29:
w00t.  Fixed it.

Specifically `rm -rf ~/.config/Code/GPUCache/*` did the trick.

So I'm willing to bet that some kind of update, like graphics drivers, caused any shaders that were compiled earlier to be invalid. And the web view cached them.

bernhard-42, 2024-04-29:
Awesome!!!
I need to write a troubleshooting section at some day - and tjis will be one of the topics for sure

bernhard-42, 2024-04-30:
Added it to the readme: https://github.com/bernhard-42/vscode-ocp-cad-viewer?tab=readme-ov-file#troubleshooting 
So I guess, we can close it

closed 2024-04-30 by bernhard-42

PaulBone, 2024-05-14:
> Added it to the readme: https://github.com/bernhard-42/vscode-ocp-cad-viewer?tab=readme-ov-file#troubleshooting So I guess, we can close it

I guess, Maybe the real problem is in vscode or the 3D rendering part for not invalidating the shaders when the driver changed?