bernhard-42/vscode-ocp-cad-viewer #74: Can't display a part more than once with 2.3.1

nesdnuma, 2024-04-28:
Hello,

I am running OCP Viewer under macOS Sonoma. I have python installed over miniforge3.

Everything was fine with the previous version. Now if I launch VSCode I can run code only once. If there's a bug in my code I start again. But if the code is correct and a part is displayed by OCP Viewer then I can't run anything anymore. I must quit and relaunch VS Code.

Also, I have this error message, I don't know if it's related: "You have deprecated linting or formatting settings for Python. Please see the [logs](command:python.viewOutput) for more details."

bernhard-42, 2024-04-28:
This error message is unrelated. linting is code analysis in the editor.
See discord for one idea to fix it.
If this doesn't work, I need more info: How do you run the code? Are there any errors in VSCode output panel? Are there any errors in VSCode developer tools (javascript). Does running the script (while VSCode viewer is still open) from the terminal work.

bernhard-42, 2024-04-30:
@nesdnuma Should be fixed with v2.3.2

closed 2024-05-04 by bernhard-42