bernhard-42/vscode-ocp-cad-viewer #75: cannot update the model after the first run (terminal hangs)

sgtkt, 2024-04-29:
Everything used to work fine a month ago, but now, after some upgrades, I can run my files just once (very similar to issue #74). It looks like the terminal is stuck after processing the file and rendering the model in the viewer. At least killing the terminal and re-running the file helps to update the model in the viewer, but brings me to the same state.
The terminal outputs 2 strange lines: "Jupyter kernel not running" and "+" before hanging. I'm not sure if they have anything to do with the issue.
I've tried to reset VSCode and reinstall the whole environment from scratch, but it hasn't changed anything.
The issue is not reproducible with running my code within a jupyter notebook instead of the py file.
The OS is Win11.
Please find more details on the attached screenshot.
[image]

bernhard-42, 2024-04-29:
I think it is the known issue with websockets. I have a fix but need to finally test and publish.
Maybe you could try to install the websockets version as described in https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/73#issuecomment-2075473803 to verify it is the same issue (it should work afterward)
Hope to publish the final fix on Wednesday

sgtkt, 2024-04-29:
The above workaround with WebSockets does fix the issue for me. Good luck with the final fix!

sgtkt, 2024-05-03:
The issue is no longer reproducible with ocp_vscode 2.3.3 and websockets rolled back to 12.0. Thanks a lot!

closed 2024-05-04 by bernhard-42