bernhard-42/vscode-ocp-cad-viewer #76: some settings are not applied to the initial state of the viewer

sgtkt, 2024-04-29:
I have recently had to reset VSCode and set up all plugins and their settings from scratch. I had to spend a lot of time and nerves trying to set up the OCP viewer, as some settings were not applied consistently.

Eventually I realized that the viewer ignores some settings **at first**, when it shows the OCP logo, but does apply them **later**, when I run my file the first time.

It's a minor issue, but it may cause a lot of confusion, so it would be nice if it's fixed someday.

[image]

bernhard-42, 2024-07-07:
I am now looking into it.
The logo is a very special object which I liked to look the same independent of any setings.

It is built with command:
```json
{
  "_splash": false,
  "ambient_intensity": 1,
  "angular_tolerance": 0.2,
  "axes": false,
  "axes0": true,
  "black_edges": false,
  "center_grid": false,
  "collapse": 1,
  "control": "trackball",
  "debug": false,
  "default_color": "#e8b024",
  "default_edgecolor": "#707070",
  "default_facecolor": "Violet",
  "default_opacity": 0.5,
  "default_thickedgecolor": "MediumOrchid",
  "default_vertexcolor": "MediumOrchid",
  "deviation": 0.1,
  "direct_intensity": 1.1,
  "explode": false,
  "glass": true,
  "grid": [true, false, false],
  "helper_scale": 1.0,
  "measure_tools": false,
  "metalness": 0.3,
  "modifier_keys": {
    "shift": "shiftKey",
    "ctrl": "ctrlKey",
    "meta": "metaKey"
  },
  "new_tree_behavior": true,
  "ortho": false,
  "pan_speed": 1,
  "render_edges": true,
  "render_joints": false,
  "render_mates": false,
  "render_normals": false,
  "reset_camera": "reset",
  "rotate_speed": 1,
  "roughness": 0.65,
  "show_parent": false,
  "show_sketch_local": true,
  "ticks": 10,
  "timeit": false,
  "tools": true,
  "transparent": false,
  "up": "Z",
  "zoom_speed": 1,
  "zoom": 0.8,
  "position": [92.4, -99.3, -6.6],
  "quaternion": [0.6047, 0.2156, 0.2603, 0.7212],
  "normal_len": 0
}
```
So, it will by intention ignore all alternative settings.

I have added this note to the next versions README:

> **Note:** Do not use the _OCP CAD Viewer_ logo to verify your _OCP CAD Viewer_ settings! The logo overwrites all your settings in VS Code with its own settings to always look the same on each instance. Use a simple own model for checking your configuration.

bernhard-42, 2024-09-03:
no further action needed

closed 2024-09-03 by bernhard-42