bernhard-42/vscode-ocp-cad-viewer #77: positioning of grids

sgtkt, 2024-04-29:
When I enable the grids, they are placed away from the origin (0, 0, 0) by 120% of the model's biggest dimension (XY and YZ to the negative half of Z and X, whereas XZ to the positive half of Y).

I understand the convenience of this default setup, but I'd like to be able to adjust their offsets from the origin (e.g. place them right at the origin), so that I can track key dimensions of my model.

Is there a way to do it? If not, would it be possible to add such parameters to the viewer screen or the settings?

[image]

bernhard-42, 2024-04-30:
This has to do with clipping. I invested a lot of time to get clipping with caps faces working when the space (with the grids as boundary) is not symmetric. But there was no way to get that working. The only way it finally worked was with the scene being symmetric around the origin or center of the object.

btw. if you click the Axes@0 button the grid toggles from being centered around the origin (0,0,0) to being centered around the object center

bernhard-42, 2024-05-04:
@sgtkt Does using Axes@0 fix your problem?

sgtkt, 2024-05-04:
Unfortunately, no, it just moves the axes, but not the grids.

I wanted to be able to place the grids anywhere I want and use them as a ruler, or at least use the XY grid at Z=0 position as the floor, same way as in the 3D printer slicer SW.

bernhard-42, 2024-05-04:
Did you try `center_grid=True` ?
```python
b = Pos(3,0,0) * Box(1, 1, 5)
show(b, center_grid=True)
```

[image]

[image]

sgtkt, 2024-05-04:
Yes, it's exactly what I wanted. It would be great if you could add this parameter to the settings, so that it can be enabled by default. Or feel free to close this issue as a duplicate of #60.

bernhard-42, 2024-05-05:
Have you tried `set_defaults(grid=(True, False, False), center_grid=True, transparent=True)` at the top of your script?

sgtkt, 2024-05-08:
Yes, I'm aware of this option, but don't want to use it because I have many independent code files, and I don't want to spoil them with code that is not relevant to their purpose and change all of them one-by-one whenever I decide to tweak something. I'm really happy that it's possible to control the visual representation and behavior of the viewer via the settings.

By the way, it'd be great if there also was a setting for 'reset_camera', so that I don't have to put it in the show() function all the time.

bernhard-42, 2024-05-09:
The split between vscode settings and `set_defaults` is by intention. Things that can break show or that are most probably model specific are only available via `set_defaults`.
How about having a config.py with `set_defaults` that gets imported into all other files?

sgtkt, 2024-05-14:
It's great that there's a way to make the viewer parameters project/model specific, and I use it sometimes, although I prefer to add such parameters in `show()` rather than `set_defaults()` (are there any differences?).

If there were no VSCode settings, I would have ended up with something like 'config.py' as a 'base' configuration. But, since we have 2 options, it makes no sense to use both at the same time, keeping some of the base parameters in VSCode settings and some in the config file.

Some settings like 'grids' or 'dark mode' can be seen more like user's personal preferences, and settings like 'measurement tools' or 'reset camera' I would just enable by default and never change further. In both cases it would be more logical to put them to the settings rather than the code.

This is just my personal feedback, and I would have no problem if you decide to keep it as it is now. There are plenty of options already.)

bernhard-42, 2024-07-07:
@sgtkt I will add `centergrid` to the workspace config, such as `grid_XY`, ... in the next release

bernhard-42, 2024-09-03:
added in 2.4.0

closed 2024-09-03 by bernhard-42