bernhard-42/vscode-ocp-cad-viewer #78: the viewer opens a duplicate instance of a jupyter file on autostart

sgtkt, 2024-04-30:
When opening VSCode with the active source file being a jupyter notebook and the viewer open side-by-side in another panel, a duplicate instance of the same jupyter file opens by itself as a new tab in the panel with the viewer.

It's a minor issue, as you can easily close the extra instance of the jupyter file, but it gets quite annoying to do it every time you open VSCode.

Here's what I can see right after opening VSCode.
[image]

bernhard-42, 2024-04-30:
Not sure that this is the viewers fault or a VS Code issue, but I can look into it

bernhard-42, 2024-07-07:
@sgtkt It was indeed a problem of OCP CAD Viewer. 
Fixed it for the next release

bernhard-42, 2024-09-03:
fixed in 2.4.0

closed 2024-09-03 by bernhard-42