bernhard-42/vscode-ocp-cad-viewer #81: Regression in latest version

MatthiasJ1, 2024-05-09:
Measure tools does not work if 1d objects are present

```py
s = Circle(1)
show_all(measure_tools=True)
```
Ok

```py
s = Circle(1)
l = Line((0,0), (1,0))
show_all(measure_tools=True)
```
The view will be blank and the Tree tab will be glitched.

[image]

Notice how the Tree tab is selected but the Material tab is displayed.

mngrm3a, 2024-05-10:
Idk if it's relevant, but this also blanks the viewer when measure tools are enabled:

```python
class APart(BasePartObject):
    def __init__(
        self,
        size: float,
        rotation: RotationLike = (0, 0, 0),
        align: Align | tuple[Align, Align, Align] = Align.CENTER,
        mode: Mode = Mode.ADD,
    ):
        part = Box(size, size, size)

        super().__init__(part, rotation, align, mode)

        RigidJoint(
            label="j1",
            to_part=self,
            joint_location=Location((0, -size / 2, 0), (90, 0, 0)),
        )

part1 = APart(5)
part2 = APart(5)
part1.joints["j1"].connect_to(part2.joints["j1"])
show_all()
```

bernhard-42, 2024-05-16:
@mngrm3a This example works for me with and without measure tools. Could it be that you had some other variable with a line left when it showed the blank view?

bernhard-42, 2024-09-03:
fixed in 2.4.0

closed 2024-09-03 by bernhard-42