bernhard-42/vscode-ocp-cad-viewer #82: showing edges blanks viewer window

mngrm3a, 2024-05-09:
Vertices and faces work without any issues but edges will blank the viewer.
```python
from build123d import *
from ocp_vscode import *

part = Box(10, 10, 10)

show_object(part)

# show_object(part.vertices() >> Axis.Z)  # works
# show_object(part.edges() >> Axis.Z)  # doesn't work
# show_object(part.faces() >> Axis.Z)  # works
```
I tested this with 2.2.2 and all 3 lines render as expected.

bernhard-42, 2024-05-09:
@mngrm3a I can only reproduce when `measure_tools`are `True`
Could it be that the issue only occurs when `measure_tools=True`? Similar as in https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/81

mngrm3a, 2024-05-10:
Indeed, it works when the measure tools are disabled.

bernhard-42, 2024-05-10:
Thanks.
Duplicate of https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/81

bernhard-42, 2024-09-03:
fixed in 2.4.0

closed 2024-09-03 by bernhard-42