bernhard-42/vscode-ocp-cad-viewer #84: Update readme with instructions that don't require mamba for macOS ARM / Apple Silicon

infused-kim, 2024-06-14:
Previously, on Apple Silicon macs the cadquery-ocp package was not installable through pip and required mamba. But now there is a wheel release on github that can be installed with pip, which means you don't need mamba anymore.

Having to use mamba was a big annoyance and deterence for me and probably others.

Visit [ocp-build-system/releases](https://github.com/CadQuery/ocp-build-system/releases) and copy the URL of the `macosx_11_0_arm64.whl` for your python version.

```bash
python3 --version

pip install https://github.com/CadQuery/ocp-build-system/releases/download/7.7.2.0/cadquery_ocp-7.7.2-cp39-cp39-macosx_11_0_arm64.whl

pip install build123d
```

For more info, you can check out the following issues:

- <https://github.com/CadQuery/ocp-build-system/issues/11>
- <https://github.com/bernhard-42/vscode-ocp-cad-viewer/issues/68>

jdegenstein, 2024-06-14:
I personally think it is better to wait for the Cadquery-OCP wheels to be on PyPI. With the approach you suggested it would require the user to pick the correct URL of the correct wheel that matches python / OS / CPU-type.

Since you are obviously aware of the workaround, then feel free to use it but IMHO this does not necessarily improve the ease-of-use for new MacOS+arm64 users.

codingandmore, 2025-01-07:
You can update the installation instructions for build3d. I installed yesterday on an Apple Silicon Mac  by just:

1. Create an environment
2. pip install build3d 
3. pip install ocp_vscode
4. install OCP CAD Viewer from VSCode Marketplace

And it worked in VSCode and standalone mode. This is so cool!!!! No conda, mamba whatever. pip and go ;)

bernhard-42, 2025-03-02:
Fixed with 2.6.2