bernhard-42/vscode-ocp-cad-viewer #86: Image export of scene possible?

MamboJambo42, 2024-07-02:
Hi,

is it possible to export the current scene of the OCP CAD Viewer as an image file? I'm really struggling here and I appreciate any help.

bernhard-42, 2024-07-14:
The next version (2.4) will have a Python command `save_screenshot(filename)`.
It will save a screenshot from the current view relative to the current working folder.
For example, for this object:

[image]

`save_screenshot("./assets/img.png")` will create:

[image]

It will not show the CAD tools nor the orientation indicator (

MamboJambo42, 2024-07-14:
This is exactly what I've been looking for! Thanks for the update, and I'm looking forward to using it soon.

bernhard-42, 2024-09-03:
Added in 2.4.0

closed 2024-09-03 by bernhard-42