bernhard-42/vscode-ocp-cad-viewer #89: Install on Code-Server

aguileraGit, 2024-08-22:
Hello.
I run [code-server](https://coder.com/) via a [Docker Image](https://docs.linuxserver.io/images/docker-code-server/). I don't see the OCP CAD Viewer listed on the extensions. A plugin for [CAD Query](https://open-vsx.org/extension/roipoussiere/cadquery) does appear.

Any reason it doesn't appear? Thoughts on installing manually?

bernhard-42, 2024-08-22:
Sorry, no idea, I don't use docker and code server

aguileraGit, 2024-08-23:
Understood. Code-Server (and Docker image) don't use the standard [Microsoft's Extension Marketplace](https://coder.com/docs/code-server/FAQ#why-cant-code-server-use-microsofts-extension-marketplace). Installing manually fixes that issue. Have you considered [Open VSX](https://github.com/open-vsx/publish-extensions)?

The second issue is lack of Python on the Docker Image. The publisher of the Docker Image has a [provision](https://docs.linuxserver.io/general/container-customization/) in place for this. I'll work through this and post any results.

bernhard-42, 2024-09-07:
Closing for now. 
If there is any issue VS Code CAD viewer specific, please reopen or open a new issue

closed 2024-09-07 by bernhard-42

binaryben, 2025-01-27:
Could you please reconsider publishing to Open VSX @bernhard-42? VSCodium also uses the Open VSX marketplace instead of Microsoft's.