bernhard-42/vscode-ocp-cad-viewer #91: ShapeList of Compound displayed incorrectly

MatthiasJ1, 2024-08-25:
```py
x = Compound(children=[Circle(1), Pos(3)*Circle(1)])
x = ShapeList([x, Pos(Y=3)*x])

show_all()
```

**Incorrect:**
[image]

```py
x, y = x # Expand list x into its components within separate variables

show_all()
```
**Correct:**
[image]

bernhard-42, 2024-08-26:
Doesn't work in the latest version yet.
I never expected someone using `ShapeList` as a wrapper to generic lists.
```python
x = Compound(children=[Circle(1), Pos(3)*Circle(1)])
x = [x, Pos(Y=3)*x]

show_all()
```
works in my latest version.

Out of curiosity, why are you doing this?

MatthiasJ1, 2024-08-27:
Well in my scenario I was projecting a sketch onto a solid which would give >1 Compounds which I was returning as a ShapeList to make it easy to sort/operate on. In this case Compounds made the most sense to me categorically but in general using unfused compounds can yield significant performance improvements.

bernhard-42, 2024-08-27:
I now differentiate between (due to internal reasons) homogenous `ShapeList`s (as returned by e.g. `edges()`) and other `ShapeList`s. The other ones will simply be treated as lists and unrolled by the converter.

bernhard-42, 2024-09-03:
fixed in 2.4.0

closed 2024-09-03 by bernhard-42