bernhard-42/vscode-ocp-cad-viewer #92: ShapeList containing different object types fails to render

MatthiasJ1, 2024-08-25:
This silently fails to render.
```py
x = ShapeList([Circle(1), Pos(2)*Sphere(1)])
show_all()
```

The list is valid, however, as the below works and renders.

```py
x = x.faces()
show_all()
```
[image]

bernhard-42, 2024-08-26:
Again, why do you use `ShapeList` for this? Why not a simple list?

MatthiasJ1, 2024-08-27:
ShapeList is just a list which also provides many CAD utility functions so whenever I have a list of CAD objects that I may want to perform queries on ShapeList > List. For example, I want my functions to return ShapeLists so I can do `project(...).faces()` rather than `ShapeList(project(...)).faces()`.

bernhard-42, 2024-08-27:
Got it, thanks
I now support generic `ShapeList`s (as opposed to the homogenous `ShapeList`s being returned by `faces()`, `edges()`, ...). It'll come with the next release in the coming days

bernhard-42, 2024-08-28:
Just posted on Discord the newest pre-release.
If you would have some time, I would really appreciate if you could give it a try - thanks :-)

MatthiasJ1, 2024-09-01:
Yes, it works now, thanks!

closed 2024-09-01 by MatthiasJ1