bernhard-42/vscode-ocp-cad-viewer #95: Add click-to-center feature

eteq, 2024-08-31:
While this tool is really amazing for build123d and cq workflows, there is one and only one feature that sometimes forces me to go back into freecad: the "click to center" feature in some of the freecad view modes.  That is, you can middle-click on a spot in the view and it re-centers the camera there, which, more importantly, means if I orbit or zoom in, it zooms into the spot I selected.  This is critical for inspecting small parts embedded in a larger part.

Right now I can of course pan to wherever I need to be with right-click, but I find that imprecise and awkward to use such that in practice I sometimes have to dump my object and load it in freecad to inspect things.  Can a middle-click-to-center feature be added to `vscode-ocp-cad-viewer` ? (or maybe this belongs better in [three-cad-viewer](https://github.com/bernhard-42/three-cad-viewer)? I confess I don't really understand the boundary of what is where)

eteq, 2024-08-31:
(Note I'm open to "well then make a PR for that!" but I am not really sure where to find the relevant interaction code, so pointers would be helpful on that...)

bernhard-42, 2024-09-01:
Everything that is in the viewer window (cad tree, canvas with CAD object(s), mouse handling, ....) is the viewer component [three-cad-viewer](https://github.com/bernhard-42/three-cad-viewer).

vscode-ocp-viewer does two things:
- Use [ocp-tessellate](https://github.com/bernhard-42/ocp-tessellate) to convert OCP to mesh (tessellation)
- transport the tessellated object from Python to VS Code and from there to the embedded viewer component (using websockets)

So the feature request you are asking about is in fact a [three-cad-viewer](https://github.com/bernhard-42/three-cad-viewer) topic as you suggested.

Currently it can be done by code:
```
set_viewer_config(target=(x,y,z))
```
[image]

bernhard-42, 2024-09-01:
And something like that can be added to the viewer component, i.e. when doing a <modifier>-click, then raytracing (already used) will be used to identify the nearest point on the object that this click would point to on an axis from the camera through the click point in the view projection point to the object. And set camera's target to this point.

The code approach above uses the measure/property mode to get the coords and then sets the target of the camera.

bernhard-42, 2024-10-19:
Fixed in 2.5.0

closed 2024-10-19 by bernhard-42