bernhard-42/vscode-ocp-cad-viewer #96: 2.4 performance bug

MatthiasJ1, 2024-09-01:
If you show a ~5MB mesh imported with `Mesher().read()` and enable measuring tools, the framerate will drop to 1 FPS (if this could be improved that would be awesome).

Next, call show with a simple object. The performance will remain bad until you restart ocp_viewer.

Also, I've noticed in this scenario I will often run into this broken empty view which requires restarting vscode to fix:

[image]

bernhard-42, 2024-09-02:
Measuring STL objects (meshes) is not supported. Most important reason: The Brep kernel face does not have edges and vertices, its a pretty strange face. On my Mac the code seg faults when trying to compute the center of an STL face.

I should mention that. I have no idea how to prevent applying measurement to objects since it can be a combination of BRep and meshes.

MatthiasJ1, 2024-09-02:
I'm not sure I follow, I am measuring a normal Brep solid. Since it is generated from an STL, it has many more faces than a CAD generated one.

bernhard-42, 2024-09-03:
I was talking about `import_stl`. The result is strange:
```python
In [46]: m = import_stl("holder.stl")

In [47]: m.solids()
Out[47]: []

In [48]: m.faces()
Out[48]: [<build123d.topology.Face at 0x28f2b1610>]

In [49]: m.edges()
Out[49]: []

In [50]: m.vertices()
Out[50]: []
```
So the viewer tessellates the face and adds the vertex normals and the triangle faces for better visualisation. These edges are not known to OCP and so backend can't measure them.

bernhard-42, 2024-09-03:
With `Mesher().read()` I can't really read STL files, it always raises different exceptions.
When I load a 3mf example (build123d's dual_color.3mf), this works in fact returns solids:
```python
In [65]: m = Mesher().read("dual_color.3mf")

In [66]: m
Out[66]: 
[<build123d.topology.Solid at 0x2cfa52610>,
 <build123d.topology.Solid at 0x2cfa50510>,
 <build123d.topology.Solid at 0x2cfa502d0>,
 <build123d.topology.Solid at 0x2cfa50250>,
 <build123d.topology.Solid at 0x2cfa50890>,
 <build123d.topology.Solid at 0x2cfa52e50>]
 ```
For this the measure mode works well.

With larger 3MFs downloaded from grabcad I typically run into 
```bash
ELib3MFException: Lib3MFException 5: Duplicate Resource ID
```
when I try to read with `Mesher`

bernhard-42, 2024-09-03:
I now created a 3mf with build123d and loaded again. Now I understand what you mean.

But this simply exceeds the limits at which OCP CAD Viewer can work: A pretty simple example with 18 solids after Mesher import blows up to >62000 faces, edges and vertices. Now, threejs can easily show millions of faces, edges and vertices, but not when each face, edge and vertex is wrapped into a separate group - which the viewer currently does to separate objects and allow for measurement (I need to have a unique id to send back the backend to do the measurement in OCP. And only groups have ids)

I guess we would need to switch to measures calculated in the viewer (which would be possible since we only have straight lines). But this needs investigation and will add a lot of complexity. 
So I don't see a fix soon, sorry.

MatthiasJ1, 2024-09-03:
Yea no worries, I get that the rendering engine struggles to handle so many individual objects. The main "bug" here is that even once the large object is no longer being shown, the bad performance remains until you restart vscode.

MatthiasJ1, 2024-09-04:
Steps to reproduce (this is for 2.3.3 which has the same bug):

1. Run the following script which generates a fancy sphere with about 3k vertices:
```py
p = Box(*[2/sqrt(3)]*3)
s = lambda p: Solid(Shell([Polygon(*[v.center() for v in e.vertices()], f.center().normalized(), align=None) for f in p.faces() for e in f.edges()]))
p = s(s(s(s(s(s(p))))))
print(f"{len(p.vertices())=}")
show_all(measure_tools=True)
```

2. Click on `Show shape properties`

3. Run the following script:
```py
p = Sphere(1)
show_all()
```

The viewer will be very laggy despite only showing a simple sphere.

bernhard-42, 2024-09-04:
Fixed in 2.4.1