bernhard-42/vscode-ocp-cad-viewer #97: 2.4 Regression: Objects hidden by object-tree in GUI reset between show calls

MatthiasJ1, 2024-09-03:
In 2.3.3 and previous versions, if you call `show_all()`, then in the tree view hid an object by clicking the eye, then call `show_all()` again, the object you explicitly hid in the GUI would remain hidden. In the new version, all objects are re-shown after each call.

```py
from ocp_vscode import *
from build123d import *
from time import sleep

a = Sphere(1)
b = Circle(2)

print("========== First show ==========")
show_all()
sleep(6)

# Here hide one of the objects in the tree view

print("========== Second show ==========")
show_all()

# in <=2.3.3 the item you hide will have remained hidden
# in =2.4.0, the item you hide will be shown
```

I typically have a lot of objects in my view but only need to reference a few at a given time.

bernhard-42, 2024-09-04:
Fixed in 2.4.1

MatthiasJ1, 2024-09-05:
Awesome, thanks!

closed 2024-09-07 by bernhard-42