bernhard-42/vscode-ocp-cad-viewer #101: Viewer window contents do not fit VSCode window

MatthiasJ1, 2024-09-21:
In the first image the XY plane grid extends past the white region.
In the second image I just scrolled to the right and the view moves off screen. The window is large enough to fit the whole view as additional spacing has been introduced to the toolbar.

[image]
[image]

bernhard-42, 2024-09-29:
Note to myself:
```css
<div id="cad_viewer" 
    style="margin-left: -20px;"
>
``` 
should solve the first issue.
The second one is more tricky. Needs more investigation

bernhard-42, 2024-10-02:
I fixed issue 1 (the grid issue)
The second issue with the black gap at the end is a strange behaviour of VS Code. The minimum cad viewer width is currently 735 to ensure all buttons fit into the tool bar. But VS Code loves 850 as minimum width (no idea why and beyond my control). So this issue can't be fixed at the moment (and additionally, I do not think it is too critical)

bernhard-42, 2024-10-19:
Issue 1 fixed in 2.5.0
Issue 2 can't be fixed at the moment, so closing for now.
Please reopen, if this is a real issue in working with the viewer

closed 2024-10-19 by bernhard-42

MatthiasJ1, 2024-10-19:
No problem, thanks for taking a look at it!