bernhard-42/vscode-ocp-cad-viewer #106: cannot run in VSCode due to numpy error

dan-p3rry, 2024-09-29:
I've tried about 15 times to install the OCP viewer in VSC .... I've tried both 3.11.1 and 3.12.3 python versions.  

This might be the same issue as your ticket #100.  Is the viewer broken, or is there a workaround?

> A module that was compiled using NumPy 1.x cannot be run in
> NumPy 2.1.1 as it may crash. To support both 1.x and 2.x
> versions of NumPy, modules must be compiled with NumPy 2.0.
> Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.
> 
> If you are a user of the module, the easiest solution will be to
> downgrade to 'numpy<2' or try to upgrade the affected module.
> We expect that some modules will need time to support NumPy 2.
> 
> Traceback (most recent call last):  File "<frozen runpy>", line 198, in _run_module_as_main
>   File "<frozen runpy>", line 88, in _run_code
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel_launcher.py", line 18, in <module>
>     app.launch_new_instance()
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\traitlets\config\application.py", line 1075, in launch_instance
>     app.start()
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel\kernelapp.py", line 739, in start
>     self.io_loop.start()
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\tornado\platform\asyncio.py", line 205, in start
>     self.asyncio_loop.run_forever()
>   File "C:\Users\danp3\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 607, in run_forever
>     self._run_once()
>   File "C:\Users\danp3\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 1919, in _run_once
>     handle._run()
>   File "C:\Users\danp3\AppData\Local\Programs\Python\Python311\Lib\asyncio\events.py", line 80, in _run
>     self._context.run(self._callback, *self._args)
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel\kernelbase.py", line 545, in dispatch_queue
>     await self.process_one()
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel\kernelbase.py", line 534, in process_one
>     await dispatch(*args)
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel\kernelbase.py", line 437, in dispatch_shell
>     await result
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel\ipkernel.py", line 362, in execute_request
>     await super().execute_request(stream, ident, parent)
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel\kernelbase.py", line 778, in execute_request
>     reply_content = await reply_content
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel\ipkernel.py", line 449, in do_execute
>     res = shell.run_cell(
>   File "C:\Users\danp3\AppData\Local\Temp\ipykernel_11704\1179718759.py", line 29, in wrapper
>     result = old_func(*args, **kwargs)
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ipykernel\zmqshell.py", line 549, in run_cell
>     return super().run_cell(*args, **kwargs)
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\IPython\core\interactiveshell.py", line 3075, in run_cell
>     result = self._run_cell(
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\IPython\core\interactiveshell.py", line 3130, in _run_cell
>     result = runner(coro)
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\IPython\core\async_helpers.py", line 128, in _pseudo_sync_runner
>     coro.send(None)
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\IPython\core\interactiveshell.py", line 3334, in run_cell_async
>     has_raised = await self.run_ast_nodes(code_ast.body, cell_name,
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\IPython\core\interactiveshell.py", line 3517, in run_ast_nodes
>     if await self.run_code(code, result, async_=asy):
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\IPython\core\interactiveshell.py", line 3577, in run_code
>     exec(code_obj, self.user_global_ns, self.user_ns)
>   File "<ipython-input-1-e552ca806f1f>", line 4, in <module>
>     from ocp_vscode import *
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ocp_vscode\__init__.py", line 22, in <module>
>     from .show import *
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ocp_vscode\show.py", line 27, in <module>
>     import ocp_tessellate.convert as oc
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ocp_tessellate\__init__.py", line 21, in <module>
>     from .cad_objects import (
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ocp_tessellate\cad_objects.py", line 7, in <module>
>     from ocp_tessellate.ocp_utils import (
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\ocp_tessellate\ocp_utils.py", line 93, in <module>
>     from quaternion import rotate_vectors
>   File "c:\Users\danp3\Documents\cadquery\.venv\Lib\site-packages\quaternion\__init__.py", line 27, in <module>
>     from .numpy_quaternion import (
>

bernhard-42, 2024-09-29:
The viewer is not broken, it still works with numpy < 2.
However, most probably cadquery depends on numpy 2. Python dependency issue.
Hence the issue 100. I don't think there is a workaround at the moment except forcefully installing numpy 1.16 I think - which might break something else.
Will try to publish a fix soon that doesn't rely on numpy-quaternion

bernhard-42, 2024-09-29:
I meant numpy 1.26

bernhard-42, 2024-09-30:
I've published [`ocp-tessellate`](https://github.com/bernhard-42/ocp-tessellate/) 3.0.5.
This package will be installed by `ocp_vscode` and had a dependency to numpy 1. 
Version 3.0.5 now works for both numpy 1 and numpy 2 and you should be able to install your environment.

**Installation:**
- To upgrade an existing environment, simply `pip install --upgrade ocp-tessellate` (or use the viewer's library manager) and then `pip uninstall numpy-quaternion`
- For new environments, the `ocp_vscode` installation will automatically install `ocp-tessellate 3.0.5` and will not install `numpy-quaternion`

closed 2024-10-01 by dan-p3rry