bernhard-42/vscode-ocp-cad-viewer #111: Config file error

Spectre5, 2024-10-19:
I used ocp_viewer a year ago or so and went to re-visit one of my models.  I created a new virtual environment and installed build123d and ocp_viewer and ran the script, but I got this exception when I tried to show the part:

```
RuntimeError: Unable to open config file /home/user/.ocpvscode.
```

Why doesn't it just make this file if it doesn't exist?  Since I don't know what to put in this file, I touched it (empty file) and ran it again, but then the show code just seems to be skipped over but some text does appear from python:

```
UserWarning: The viewer doesn't seem to run
Jupyter console not installed
```

However, the viewer is clearly running on the right side of Codium and the extension clearly indicates that it is installed:

[image]

I then saw a message from the extension in Codium in the bottom right that shows:

[image]

So, I added `set_port(3939)` at the top of my code and ran it again.  Now the message above is not shown (the UserWarning), but I still don't see anything in the ocp_code viewer window.

I don't see any mention of this config file in the README, so is there something else I need to do here?

bernhard-42, 2024-10-19:
You seem to still have an older version of the extension (2.0.13), the newest is 2.5.0. And from the warning message you are running 2.5.0 of the Python code.
Lot's of things have changed since then and both versions need to be aligned (which happens when you let the VS code extension install the Python module). 
Please install the latest version of the extension and try again

Spectre5, 2024-10-20:
I'm used to the extensions auto-updating, but I guess since I use Codium and had to do the manual install that it obviously doesn't auto-update.  Updating via the latest manual install seems to have fixed the issue - thanks!

closed 2024-10-20 by Spectre5