bernhard-42/vscode-ocp-cad-viewer #112: OCP Viewer stays empty after first install

CePeU, 2024-10-21:
After a fresh install on Windows 10 (x86) after opening a "build123d.py" file the first time the OCP Viewer stays empty. The workaround is to close the OCP Viewert and save the ".py" file.

First install:

[image]

X'ing welcome window:

[image]

Checking installed plugins and build123d environment:

[image]

[image]

Loading first .py  file --> OCP CAD Viewer is empty:

[image]

After closing the OCP CAD Viewer with X and saving the BeeBox.py OCP CAD Viewer ist correctly initialized:

[image]

Please note that with version 2.5.0 I also get an additional message that the syntax for the path name is wrong. 

"Die Syntax für den Dateinamen, Verzeichnisnamen oder die Datenträgerbezeichnung ist falsch. -->
"The syntax for the file name, directory name or data carrier designation is incorrect."

The only difference seem to be the additional hypens the drive paths are now encased in. See picture below foOCP Cad Viewer 2.4.1

[image]

CePeU, 2024-10-22:
Ok I started an installed viewer and it worked and left a . ocpvscode file which was filled.
{
  "3939": {
    "roots": []
  }
}
I then copied the file and deleted the .ocpvscode file.
I deleted . ocpvscode and did a full install of the portable  setup. During setup phase a new totally empty 0 byte .ocpvscode was generated.
I overwrote that empty file with the copied one above.

I opened VsCode, I checked for the extensions and Xed the welcome screen.

I opened up a python file and I got a stuck viewer:
[image]

It tells me it started the backend viewer and also on which port.

THEN I tried to run the python file just to see what happens, and I get this output:

[image]

It  claims it took the port from the config (and this seems to be true) but it also claims it could not access the viewer  config. Is there one more config file?

bernhard-42, 2024-10-26:
I unfortunately have no idea what happens here. I did several fresh installations with your installer and no place in VS Code showed any error. Since I do not understand what this portable setup does under the hood, I can only give up, sorry.

How about removing VS Code and using the standalone viewer?
User only needs to open the WinPython Terminal and type `python -m ocp_vscode` and open the browser?
Not suer it is better because then they need to choose an editor which respects the WinPython environment.
So again, no real idea ...

bernhard-42, 2025-03-02:
Closing for now

closed 2025-03-02 by bernhard-42