bernhard-42/vscode-ocp-cad-viewer #115: Certain faces cannot be inspected

MatthiasJ1, 2024-10-25:
In this example, the cylindrical faces cannot be inspected.

```py
t1 = SlotOverall(10, 5)
t2 = Pos(Z=2) * SlotOverall(8, 3)
t = loft([t1, t2])
t = offset(t, 1)
```

[image]
[image]

bernhard-42, 2024-10-26:
Note to myself:
According to the backend Terminal output:
```
response.radius = shape._geom_adaptor().Cylinder().Radius()
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'OCP.Geom.Geom_RectangularTrimmedSurface' object has no attribute 'Cylinder'
```

check out which results `shape._geom_adaptor()` could have and handle them

bernhard-42, 2024-10-26:
https://dev.opencascade.org/doc/refman/html/classGeom__Surface.html
[image]

bernhard-42, 2024-11-07:
Fixed in 2.6.1

closed 2024-11-07 by bernhard-42