bernhard-42/vscode-ocp-cad-viewer #116: Unable to visualize "simple" circle sketch

CasperH2O, 2024-11-03:
Found this one when porting my prototypes/tests from CQ-Editor to OCP CAD Viewer. Not sure what I'm doing wrong, this is what is shown in CQ Editor with this code:

[image]

Traceback:

```
Traceback (most recent call last):
  File "c:\Users\caspe\PycharmProjects\3DMarbleMaze\3DMarbleMazeGenerator\shapes\tests\path_sweep_single-cq-ocp-viewer-compare.py", line 26, in <module>
    show_object(circle_profile_1)
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_vscode\show.py", line 774, in show_object
    show(
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_vscode\show.py", line 551, in show
    t, mapping = _convert(
                 ^^^^^^^^^
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_vscode\show.py", line 251, in _convert
    instances, shapes, config, count_shapes, mapping = _tessellate(
                                                       ^^^^^^^^^^^^
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_vscode\show.py", line 213, in _tessellate
    instances, shapes, mapping = tessellate_group(
                                 ^^^^^^^^^^^^^^^^^
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_tessellate\convert.py", line 1594, in tessellate_group
    mapping, shapes = group.collect(
                      ^^^^^^^^^^^^^^
    mapping, mesh = obj.collect(
                    ^^^^^^^^^^^^
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_tessellate\cad_objects.py", line 140, in collect
    values, bb = convert(self.obj, self.name, self.id)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_tessellate\convert.py", line 1581, in _discretize_edges
    disc_edges = discretize_edges(obj, deflection, id_)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_tessellate\tessellator.py", line 531, in discretize_edges
    edge_types.append(get_edge_type(edge).value)
                      ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\caspe\cq-editor\Lib\site-packages\ocp_tessellate\ocp_utils.py", line 607, in get_edge_type
    return BRepAdaptor_Curve(edge).GetType()
           ^^^^^^^^^^^^^^^^^^^^^^^
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. OCP.BRepAdaptor.BRepAdaptor_Curve()
    2. OCP.BRepAdaptor.BRepAdaptor_Curve(E: OCP.TopoDS.TopoDS_Edge)
    3. OCP.BRepAdaptor.BRepAdaptor_Curve(E: OCP.TopoDS.TopoDS_Edge, F: OCP.TopoDS.TopoDS_Face)

Invoked with: <OCP.TopoDS.TopoDS_Vertex object at 0x000002227A96FC30>
```

Minimal code to reproduce:

```
import cadquery as cq
from ocp_vscode import *

# Create a tube profile (the inner and outer circles)
diameter_inner = 8
diameter_outer = 10 - 0.001

# Profile for part 1
circle_profile_1 = (
    cq.Workplane("ZY")
    .workplane(offset=50)
    .circle(diameter_outer / 2)
    .circle(diameter_inner / 2)
    .consolidateWires()    # Also tried without this one
)

show_object(circle_profile_1)
```

bernhard-42, 2024-11-07:
Fixed in 2.6.1

closed 2024-11-07 by bernhard-42

CasperH2O, 2024-11-07:
Can confirm, thank you @bernhard-42 . 👍