bernhard-42/vscode-ocp-cad-viewer #119: Generic show

adam-urbanczyk, 2024-11-14:
Hey, would you consider working on/with https://github.com/CadQuery/cadquery/issues/1684 ? IDK yet what the architecture of it should be, but it would be nice to be able to have visualization-layer agnostic code.

bernhard-42, 2024-11-16:
Sounds like an interesting idea.

OCP CAD Viewer is CAD lib agnostic, i.e. it has the same usage across all supported code cad systems.
There is (Cadquery)
```python
import cadquery as cq
from ocp_vscode import show, set_defaults

b = cq.Workplane().box(1,1,1)

show(b)
show(b.val().wrapped)
```
and (Build123d)
```python
from build123d import *
from ocp_vscode import show, set_defaults

b2 = Box(1,1,1) # Build123d 

show(b2)
show(b2.wrapped)
```
and (Build123d / CadQuery integration)
```python
from build123d import *
import cadquery as cq
from ocp_vscode import show, set_defaults

b = cq.Workplane().box(1,1,1) # CadQuery

b3 = Solid(b.val().wrapped)
show(b3)
```
and (pure OCP)
```python
from OCP.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCP.gp import gp_Ax2
from ocp_vscode import show, set_defaults

b4 = BRepPrimAPI_MakeBox(gp_Ax2(), 1, 1, 1).Shape()

show(b4)
```

So, if we would have something like matplotlib
```python
import cadquery as cq
cq.use("ocp_vscode")
```
it would be a different approach than for build123d and OCP.

So a different idea could be to agree on the exported functions and their behaviour.

```python
from ocp_vscode import show, show_object, show_all, set_defaults
from cq_editor import show, show_object, show_all, set_defaults
from cadquery.vis import show, show_object, show_all, set_defaults
```

There could be a new function `start_viewer()` that all three expose:
- OCP CAD Viewer would start its standalone viewer (and if possible start a browser opening the viewer URL)
- CQ Editor would start its QT viewer 
- cadquery.vis would start the VTK based viewer

That looks very portable to me.
What would be the advantage of a generic backend interface

adam-urbanczyk, 2024-11-16:
So is it a no then or am I misunderstanding your anwser? To be clear, my proposal would be to provide some kind of backed registration mechanism and on your side you'd need to register it when importing/executing CQ code. This way the following code:
```python
import cadquery as cq
from cadquery.vis import show

show(cq.Workplane().box(1,1,1))
```
would behave differently depending on where it is run (i.e. vanilla interpreter, CQ-editor or VS-Code). The added value for the users would be that they don't need to modify their code depending on where they want to run it.

bernhard-42, 2024-11-16:
It is not a no atm, still trying to understand the benefit.
How would the registration of ocp_vscode work?
In matplotlib I have to add a line `matplotlib.use("agg")` for example.

bernhard-42, 2024-11-16:
`ocp_vscode` always works on a vanilla interpreter, whether as standalone or in VS Code.
The Python part (code CAD code) and the viewer are fully decoupled. The viewer (both standalone and in VS Code) start a websocket server and `show` communicates with it.
Given that you always have to import `ocp_vscode`

bernhard-42, 2024-11-16:
The standalone viewer can even be on a remote machine, btw.
So there is no "OCP CAD Viewer" environment as with CQ Editor.
As every other standard Python package you install it with `pip install` and then import it.
Hence I don't fully how your idea would work in case of OCP CAD Viewer

adam-urbanczyk, 2024-11-16:
OK, never mind. It does not sound like a feasible idea for OCP Viewer.

closed 2024-11-16 by adam-urbanczyk