bernhard-42/vscode-ocp-cad-viewer #121: Eliminate requirement for `set_port(3939)` in "standalone" mode

jdegenstein, 2024-11-18:
When running in "standalone" mode (i.e. not using VSCode) and `python -m ocp_vscode` I get the following error when `set_port(3939)` is not explicitly included:
```py
In [1]: %run b123d_ocp-vscode.py
Lock file /home/user/.ocpvscode.lock already exists
Lock file /home/user/.ocpvscode.lock is stale
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/state.py:156, in get_state()
    155 try:
--> 156     with open(config_file, "r", encoding="utf-8") as fd:
    157         config = fd.read()

FileNotFoundError: [Errno 2] No such file or directory: '/home/user/.ocpvscode'

The above exception was the direct cause of the following exception:

RuntimeError                              Traceback (most recent call last)
File ~/Temporary/b123d_ocp-vscode.py:23
     20     mirror(ex16_single.part, about=Plane.YZ.offset(width))
     21     mirror(ex16_single.part, about=Plane.YZ.offset(-width))
---> 23 show(ex16.part)

File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/show.py:555, in show(names, colors, alphas, port, progress, glass, tools, tree_width, axes, axes0, grid, ortho, transparent, default_opacity, black_edges, orbit_control, collapse, explode, ticks, center_grid, up, zoom, position, quaternion, target, reset_camera, clip_slider_0, clip_slider_1, clip_slider_2, clip_normal_0, clip_normal_1, clip_normal_2, clip_intersection, clip_planes, clip_object_colors, pan_speed, rotate_speed, zoom_speed, deviation, angular_tolerance, edge_accuracy, default_color, default_edgecolor, default_facecolor, default_thickedgecolor, default_vertexcolor, ambient_intensity, direct_intensity, metalness, roughness, render_edges, render_normals, render_mates, render_joints, show_parent, show_sketch_local, helper_scale, mate_scale, debug, timeit, _force_in_debug, *cad_objs)
    552 progress = Progress([] if progress is None else [c for c in progress])
    554 with Timer(timeit, "", "overall"):
--> 555     t, mapping = _convert(
    556         *cad_objs,
    557         names=names,
    558         colors=colors,
    559         alphas=alphas,
    560         progress=progress,
    561         **kwargs,
    562     )
    564     if not _force_in_debug:
    565         LAST_CALL = "show"

File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/show.py:255, in _convert(names, colors, alphas, progress, *cad_objs, **kwargs)
    252 if progress is None:
    253     progress = Progress([c for c in "-+c"])
--> 255 instances, shapes, config, count_shapes, mapping = _tessellate(
    256     *cad_objs,
    257     names=names,
    258     colors=colors,
    259     alphas=alphas,
    260     progress=progress,
    261     **kwargs,
    262 )
    264 if config.get("dark") is not None:
    265     config["theme"] = "dark"

File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/show.py:86, in _tessellate(names, colors, alphas, progress, *cad_objs, **kwargs)
     83 def _tessellate(
     84     *cad_objs, names=None, colors=None, alphas=None, progress=None, **kwargs
     85 ):
---> 86     if workspace_config().get("_splash"):
     87         conf = combined_config(use_status=False)
     88         reset_camera = Camera.RESET

File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/config.py:431, in workspace_config(port)
    423     return {
    424         "_splash": False,
    425         "default_facecolor": (1, 234, 56),
    426         "default_thickedgecolor": (123, 45, 6),
    427         "default_vertexcolor": (123, 45, 6),
    428     }
    430 if port is None:
--> 431     port = get_port()
    432 try:
    433     conf = send_command("config", port=port)

File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/comms.py:111, in get_port()
    108     return 3939
    110 if not INIT_DONE:
--> 111     find_and_set_port()
    112     set_connection_file()
    113 return CMD_PORT

File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/comms.py:302, in find_and_set_port()
    300     print(f"Using predefined port {port} taken from environment variable OCP_PORT")
    301 else:
--> 302     port = find_port()
    303     if port is not None:
    304         print(f"Using port {port} taken from config file")

File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/comms.py:266, in find_and_set_port.<locals>.find_port()
    264 port = None
    265 current_path = Path.cwd()
--> 266 states = get_state().items()
    267 for p, state in states:
    268     if not port_check(int(p)):

File ~/miniconda3/envs/b123d_nvim/lib/python3.11/site-packages/ocp_vscode/state.py:164, in get_state()
    162     unlock(config_file)
    163 except FileNotFoundError as ex:
--> 164     raise RuntimeError(f"Unable to open config file {config_file}.") from ex
    166 return config

RuntimeError: Unable to open config file /home/user/.ocpvscode.
```

Not sure if it is possible to eliminate this problem, but wanted to raise the issue so it is not forgotten.

rawwerks, 2025-02-11:
i'm seeing this same error...but the port 3939 trick isn't fixing it...

```
    viewer_process = subprocess.Popen([
        "python", "-m", "ocp_vscode",
        "--port", "3939",
        "--host", "127.0.0.1",
        "--debug",
        "--create_configfile"
    ])
```

gives this when i execute code:
```
{"stdout": "", "stderr": "Traceback (most recent call last):\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/state.py\", line 156, in get_state\n    with open(config_file, \"r\", encoding=\"utf-8\") as fd:\nFileNotFoundError: [Errno 2] No such file or directory: '/root/.ocpvscode'\n\nThe above exception was the direct cause of the following exception:\n\nTraceback (most recent call last):\n  File \"<string>\", line 18, in <module>\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/show.py\", line 558, in show\n    t, mapping = _convert(\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/show.py\", line 258, in _convert\n    instances, shapes, config, count_shapes, mapping = _tessellate(\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/show.py\", line 87, in _tessellate\n    if workspace_config().get(\"_splash\"):\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/config.py\", line 431, in workspace_config\n    port = get_port()\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/comms.py\", line 111, in get_port\n    find_and_set_port()\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/comms.py\", line 302, in find_and_set_port\n    port = find_port()\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/comms.py\", line 266, in find_port\n    states = get_state().items()\n  File \"/usr/local/lib/python3.10/site-packages/ocp_vscode/state.py\", line 164, in get_state\n    raise RuntimeError(f\"Unable to open config file {config_file}.\") from ex\nRuntimeError: Unable to open config file /root/.ocpvscode.\n", "exit_code": 1}%  
```

bernhard-42, 2025-02-11:
It is known and will be fixed in the next release.
For the time being, just remove the `--create-configfile` and create a file `/root/.ocpvscode` containing `{}`.
There is a major bug that I create the right config file, but them try to use the one from the VSCode version of the viewer

rawwerks, 2025-02-12:
thank you, i will give that a try!

bernhard-42, 2025-03-02:
Fixed with 2.6.2

closed 2025-03-02 by bernhard-42