bernhard-42/vscode-ocp-cad-viewer #124: Low priority feature request: add flag to remove XYZ axes from bottom left corner of save_screenshot

MatthiasJ1, 2024-11-25:
Personally I feel like saving just the shown objects should be the default behavior. I was surprised to see the additional elements.

bernhard-42, 2024-11-26:
Normally, one would have one canvas for the viewer and one canvas for the xyz axes.
However in webGL this means 2 WegGL contexts of a typical maximum of 16 in the browser. Since the viewer component also runs in jupyterlab, where each cell could have a different viewer, I combined cad view and xyz view into one canvas (threejs can define rectangles, i.e view ports, to show the right stuff).

Getting a screenshot in threejs means triggering `update` to draw the latest into the buffers and then store the canvas buffer as image. Since the xyz axes are part of the same canvas, they will be stored, too.

I think I can make the xyz axes and labels transparent before taking screenshot and then show them again afterwards. Need to check.

bernhard-42, 2025-03-02:
Fixed with 2.6.2

closed 2025-03-02 by bernhard-42