bernhard-42/vscode-ocp-cad-viewer #128: feature request: ability to put a shape name in clipboard

roipoussiere, 2024-12-09:
In my personal experience, one of the most difficult thing when moving from a GUI-based CAD software like FreeCAD or SolidWorks to a CAD library is to select a specific shape (face, edge or point).

For instance to make a fillet with a CAD GUI, we typically select an edge then we click on a fillet button. With code, we have functions to create fillets but selecting the right edge on a complex object is whole different ballgame.

Maybe we can improve this by mixing the gui-way and the cli-way, in two steps:
1. make it easier to get the shape name (those given by the *Show shape properties* feature, such as `Group/cb/faces/faces_1`) from the viewer;
2. add functions to select a face in the cad library by its name (such as `cb.get_face('faces_1')` or `Face.by_name('Group/cb/faces/faces_1')`).

Step 1. belongs to this project, so here is my proposal: allow the user to easily put the shape name (which could be a face, an edge or a point) in the clipboard. It can be done via a context menu, and/or keyboard shortcuts, and/or buttons in the toolbar.

bernhard-42, 2024-12-10:
Sure, this could be done, but is most probably useless.
OCCT does not work with fixed ids for objects. If you have square and a triangle, and you fuse them, then the new edges will have a new order and new ids. If you chamfer a square, you get 4 more edges and all edges are new and most probably in a new order.

In GUI CAD you select an object and apply an action. It is very incremental.
Code runs the same code again an again, and just because it was edge 17 in the last run, doesn't mean it is edge 17 in the next run, Fully depends on what you do after Ou "found" edge 17.

The viewer doesn't have edge IDs because OCCT doesn't. 
So I don't think it is worth spending time.

You can try it yourself:
```
b = Box(1,2,3)
show(*b.faces())
```
Now all faces are numbered in the viewer. This is enumeration is stable across runs.
The top faces has number 6.
Now do
```
b = chamfer(b, 0.2)
show(*b.faces())
```
Now the top face has number 11.

roipoussiere, 2024-12-10:
Ok, thank you for the explanation.

closed 2024-12-10 by roipoussiere