bernhard-42/vscode-ocp-cad-viewer #130: feature proposal: objects navigation

roipoussiere, 2024-12-11:
It could be useful to quickly navigate between objects, for instance to get the history of a construction.

Note that it differs with the Tree tab:
- not nested;
- one click to show an other object and hide the previous one (with eventually a quick navigation by holding click while moving cursor over several items);
- semantics: objects navigation will be typically be used to show versions of an object, while the Tree shows nested parts of an object;
- can be used in conjunction with the tree to show/hide some parts of each objects.

Let's suppose we have an object created in 3 steps (here with Build123d):

```py
obj1 = Box(3, 3, 1)
obj2 = chamfer(obj1.edges()[0], 0.5)
obj3 = obj2 - Cylinder(1, 1)
```

We could show the navigation tab with:

```py
show_objects([
    (obj1, 'obj1'),
    (obj2, 'obj2'),
    (obj3, 'obj3'),
])
```

With a helper function to get shapes from global variables, it could be shortened with:

```py
show_objects(get_globals(type=Part, prefix='obj'))
```

This way we can filter which objects to show, instead of showing all displayable object like `show_all()` .

Also when using other `show` functions, the UI could remain unchanged.

Mockup

[image]

bernhard-42, 2024-12-11:
I am not sure where the object versions should come from?
Each time you show an object, the current object will be tessellated (i.e. the BRep format will be transformed to a mesh structure suitable for threejs). There is no info about any versions.

bernhard-42, 2024-12-11:
The navigation tree is meant to navigate between objects. With modifier keys (see help dialog) you can click an object to hide or to isolate at the center

roipoussiere, 2024-12-11:
> I am not sure where the object versions should come from?

I used the word *versions* because it was more correct in a semantic point of view, but it's actually regular objects.

Thank you.

> The navigation tree is meant to navigate between objects. With modifier keys (see help dialog) you can click an object to hide or to isolate at the center

Nice! But when I shift-click on a tree element, the element is expanded, so root elements are not quickly accessible, because I need to collapse the tree first.

Is there a solution to always keep tree elements collapsed (ie. lock the *Expand root node only* button)?

bernhard-42, 2024-12-12:
> Is is possible to always keep tree elements collapsed (ie. lock the Expand root node only button), or to isolate elements without expanding them?

no

roipoussiere, 2024-12-12:
> no

Ok. I'm closing this in favor of #132, then. I think this is a better approach.

closed 2024-12-12 by roipoussiere