bernhard-42/vscode-ocp-cad-viewer #131: The help dialog might overlap the help button

roipoussiere, 2024-12-11:
The help dialog is positioned on the bottom-left corner, so if the help dialog height is bigger than the web view height, the dialog overlap the button dialog, making it difficult to close the dialog.

[image]

Possible solution: move the dialog at the left of the help button, so the button is always accessible.

bernhard-42, 2024-12-12:
pressing ESC also closes the dialog.
What screen size do you have?

roipoussiere, 2024-12-12:
> pressing ESC also closes the dialog.

It's the first thing I tried and it didn't work. Apparently ESC have no effect if you clicked on the help dialog (which I did unintentionally).

But even if it's fixed I think it's better to be able to close it with the pointer. An other solution is to close the dialog when you click outside it.

> What screen size do you have?

1920x1200, but I set VSCode zoom level to 2 (+40%).

bernhard-42, 2025-03-02:
Fixed with 2.6.2

closed 2025-03-02 by bernhard-42